stereolabs/zed-sdk #1: ErrorCode: Calibration File Not Available while using Zedfu App

sumitjagdale, 2017-04-09:
Hello,
I have installed new sdk v2, just curious about spatial mapping app within zedfu. I have already calibrated Zed after installation, rather my calibration file exists within following directory  "C:\Users\user\AppData\Roaming\Stereolabs\settings\SN1899".
But when I tried accessing this app for live as well as offline mode I am getting following error message(Attached snapshot).
Please help. Many thanks in advance.!
[image]

sl-braun, 2017-04-10:
Hi,

You need to make sure that you have access to the folder Roaming/stereolabs/settings/ without having to type a password. 
Make sure that <user> matches the loggin session (don't log in PUBLIC for example).

In any case, I would recommend to contact the support so that they can make you run a diagnostic.

Best,
/*OB*/

sumitjagdale, 2017-04-11:
@sl-braun Dear Oliver,
Thank you very much for your feedback. As I could not figure out the issue, simply reinstalled it and it has solved my problem:) 
Best,
Sumit

closed 2017-04-11 by sumitjagdale