stereolabs/zed-sdk #3: libpng error when converting or playing pack .svo file

Eirtrand, 2017-04-14:
Hi, Im currently stressed out because some important footage turned out to be incomplete and useless. I was recording using the svo recorder and saving everything to a memstick. When I run the svo file later I get several of these messages: 
libpng error: Not a PNG file
libpng error: Not a PNG file
libpng error: Not a PNG file
libpng error: Not a PNG file

When I convert the file I just get 6 messages on a 40s video, and running the svo file with the ros wrapper gives me a constant stream of the messages. It looks like the file gets corrupted or something. Im using the TX1 and starting/stopping the recording through ssh (not stable connection, loosing connection when I go underwater). The memstick is connected to a usb2 port and Im recording in 720p.

sl-braun, 2017-04-14:
Hi,

You can first try to see if the file is indeed corrupted : 

ZED SVOEditor -inf <sovfile.svo> will give you informations about the SVO file (resolution, number of frames...). Check that the values are coherent.

Open the file with ZED Explorer to check if you can get the images from the SVO.

Best,
OB

Eirtrand, 2017-04-14:
Info did not give me much information: 

Image Size : [ 1280  x  720 ]
Framerate :  30
Number of Frames :  443
ZED Serial Number :  12132

The video works, I get the same result as with the conversion, an incomplete video with a lot of footage missing. It starts ok for he first 3 seconds, then skips 5 seconds and gives me another 4 sek before ending about 15 sek short from where I stopped the recording. I replicated the issue later and got a similar result.

sl-braun, 2017-04-14:
The SVO file is not corrupted then since SVOEditor gives you correct informations.
The thing is that you have a lot of corrupted frames since you are recording on a USB2 stick (not enough bandwith to store the data.

The best would be to first use the LOSSY compression (less data) and store the svo file on a SD card (with Jetson you can put a SD card) that has a good writing/reading speed (I mean designed for recording and not storage only)

Best,
OB

Eirtrand, 2017-04-14:
I just confirmed your answer, it was indeed the USB stick. What a real bummer, I will use a sd-card from now on. Thank you for the quick and accurate support, its strongly appreciated!

closed 2017-05-19 by nesnes