stereolabs/zed-sdk #9: Multi-camera support for Windows

WASCHMASCHINE, 2017-05-31:
Hello, 

API of InitParameters -- int camera_linux_id
"**Currently** not available for Windows"

For my application it would be really useful to use 2 or more ZED cameras under Windows. Most VR and AR devices are Windows-only, which means I need a Ubuntu machine to record my videos and a Windows machine for the display, which introduces delay. I implore you to give us an update - is there still hope or did you abandon this feature completely?

adujardin, 2017-05-31:
Hi,

We didn't abandon this feature. 
However, it requires quite a lot of work. Basically, we need to change the way we communicate with the camera on Windows, which isn't as easy as on Linux. Unfortunately, it is not our current priority. 

I can keep you informed if it's scheduled for a ZED SDK release, but it might be in several months.

WASCHMASCHINE, 2017-06-01:
Hi adujardin,

thank your for your reply! Please keep me posted when it is scheduled!

adujardin, 2018-04-13:
@philomathikus Yes actually!
The multi-camera support for Windows will be included in the new version of the ZED SDK (2.4.0) alongside a new firmware (v1523) for the camera.
You could expect a beta soon.

philomathikus, 2018-04-13:
hi @adujardin, thanks for the quick reply!

We are trying to use the Zed Camera for a MR application in a UWP build but we are getting the following error only when we try to run the app after installation:

Exception thrown: 'System.DllNotFoundException' in Assembly-CSharp.dll'ZedTest.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Zed UWP\ZedTest\bin\x64\Release\AppX\System.Diagnostics.StackTrace.dll'. Cannot find or open the PDB file.DllNotFoundException: Unable to load DLL 'sl_unitywrapper': The specified module could not be found. (Exception from HRESULT: 0x8007007E)at sl.ZEDCamera.dllz_register_callback_debuger(DebugCallback callback)at sl.ZEDCamera.GetInstance()at ZEDManager.Awake()at ZEDManager.$Invoke7(Int64 instance, Int64* args)at UnityEngine.Internal.$MethodUtility.InvokeMethod(Int64 instance, Int64* args, IntPtr method)(Filename: <Unknown> Line: 0)

so everything works fine in editor.

WASCHMASCHINE, 2018-04-13:
@adujardin Great news!!

obraun-sl, 2018-04-13:
Hi,

@philomathikus :
The ZED unity wrapper (sl_unitywrapper.dll)  links with sl_zed64.dll, sl_input64,.dll sl_core64.dll and sl_mr_core64.dll. Make sure that those dlls are in the system path or in the same folder as sl_unitywrapper.dll.
When you  build an app on Unity, sl_unitywrapper.dll and sl_mr_core64.dll are copied in the <*>_data folder.

adujardin, 2018-04-17:
Hi, 
If you want to try the multi-camera on Windows, you can download an RC version of the ZED SDK 2.4.0 available at http://download.stereolabs.com/zedsdk/2.4/RC

Please keep in mind that it's still an RC version and therefore may contain bugs. Feel free to contact us if you find something at support@stereolabs.com

closed 2018-06-01 by adujardin