stereolabs/zed-sdk #11: broken installer

dtmoodie, 2017-07-02:
./ZED_SDK_Linux_Ubuntu16_v2.0.1.run 
Verifying archive integrity... All good.
Uncompressing 'ZED camera SDK by Stereolabs'  100%  
To continue you have to accept the EULA. Accept  (Y/N) ? 
Installing...
Installation path: /usr/local/zed
Checking CUDA version...
OK: Found CUDA 8.0
Do you want to auto-install dependencies (recommended) ? following packet will be installed via apt-get : libqt5sql5-mysql libqt5svg5 libqt5gui5 libqt5widgets5 libqt5opengl5 unixodbc-dev libpq5 libsybdb5 libglew-dev ttf-mscorefonts-installer libeigen3-dev freeglut3-dev libpng12-dev libboost-all-dev libturbojpeg libjpeg-turbo8 curl (Y/N) ? 
Auto-installing Tools dependencies
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package ttf-mscorefonts-installer is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ttf-mscorefonts-installer' has no installation candidate
cp: cannot stat '/usr/local/zed/tools': Permission denied
mkdir: cannot create directory '/usr/local/zed': Permission denied
mkdir: cannot create directory '/usr/local/zed': Permission denied
cp: failed to access '/usr/local/zed/lib/': Permission denied
cp: failed to access '/usr/local/zed/include/': Permission denied
cp: cannot stat '/usr/local/zed/zed-config-version.cmake': Permission denied
cp: cannot stat '/usr/local/zed/zed-config.cmake': Permission denied
Install samples (recommended) (Y/N) ? 
Installation path: /usr/local/zed/sample/
mkdir: cannot create directory '/usr/local/zed': Permission denied
cp: failed to access '/usr/local/zed/sample//': Permission denied
mkdir: cannot create directory '/usr/local/zed': Permission denied
cp: failed to access '/usr/local/zed/plugins/': Permission denied
mkdir: cannot create directory '/usr/local/zed': Permission denied
mkdir: cannot create directory '/usr/local/zed': Permission denied
cp: failed to access '/usr/local/zed/firmware/': Permission denied
cp: cannot stat '/usr/local/zed/doc': Permission denied
It appears that no graphical sessions are running, this is known to cause issues with the ZED SDK
ZED SDK installation complete !
developer@blastoise:~/code> sudo ls /usr/local/zed -alh
total 8.0K
drwx------  2 root root 4.0K Jul  2 17:15 .
drwxr-xr-x 64 root root 4.0K Jul  2 17:15 ..

WASCHMASCHINE, 2017-07-06:
Absolutely not a linux guy, but have you tried sudo for the initial command / run file?

santonatos, 2017-07-14:
try running the coammand with 'sudo' before? like  : 
sudo ./ZED_SDK_Linux_Ubuntu16_v2.0.1.run

fedario, 2018-05-14:
The problem is that the installation script makes some assumptions about the system that are problematic. Namely:

`line 214: sudo chown "${USER}" -R "${installation_folder}"`
`line 341: sudo cp -r "${folder}/99-slabs.rules" "/etc/udev/rules.d/"`

Those lines are in the linux_install_release.sh file which is inside the .run file.

If you are trying to install the run file on a system where the $USER variable is not defined you will not be able to install the SDK properly. The standard Ubuntu 16.04 docker image is one example where this won't work. The second line has a similar issue.

A workaround would be to set the $USER variable and create the /etc/udev/rules.d directory manually beforehand:
`export USER=_insert_user_`
`sudo mkdir -p /etc/udev/rules.d/`

darkdragon-001, 2018-05-16:
I am running this inside the standard Ubuntu 16.04 docker image.

I can't successfully run the run scipt via

    ./ZED_SDK_Linux_Ubuntu16_v2.0.1.run

or

    sudo ./ZED_SDK_Linux_Ubuntu16_v2.0.1.run

because it results in

    Verifying archive integrity... All good.
    Uncompressing 'ZED camera SDK by Stereolabs'  100%  
    ./ZED_SDK_Linux_Ubuntu16_v2.4.0.run: 1: eval: ./linux_install_release.sh: Permission denied

When I extract it manually via

    ./ZED_SDK_Linux_Ubuntu16_v2.4.0.run --noexec --keep --target zed && cd zed

I find the missing `./linux_install_release.sh`. I can then continue with the method described by @fedario.

fedario, 2018-05-17:
You can also do this:

`
# updating Ubuntu repository
RUN apt update

# apt package installation
RUN apt install -y sudo apt-utils udev build-essential cmake git libqt5xml5 libopencv-dev python-opencv wget curl unzip
RUN DEBIAN_FRONTEND=noninteractive apt install -y libqt5svg5 libqt5gui5 libqt5widgets5 libqt5opengl5 unixodbc-dev libsybdb5 libglew-dev ttf-mscorefonts-installer freeglut3-dev libpng12-dev libturbojpeg libjpeg-turbo8 curl libhidapi-dev libhidapi-libusb0

# fixes an issue in the ZED SDK installation below
RUN mkdir -p /etc/udev/rules.d/

# setting up a new user since the ZED SDK .run file doesn't work when run as root
RUN adduser --disabled-password --gecos '' appuser
RUN adduser appuser sudo
RUN echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers
RUN usermod -a -G video appuser
USER appuser
ENV USER appuser

# ZED SDK installation
COPY ZED_SDK_Linux_Ubuntu16_v2.4.0.run /home/appuser
RUN sudo chmod +x /home/appuser/ZED_SDK_Linux_Ubuntu16_v2.4.0.run
RUN /home/appuser/ZED_SDK_Linux_Ubuntu16_v2.4.0.run -- silent
`

willcbaker, 2018-06-26:
seems the installer also defaults to a /usr/local but the permissions are still tied to the user; can we set group permissions to these files instead?

closed 2021-12-01 by bot-stereolabs