stereolabs/zed-sdk #12: zed- camera control / opencv

santonatos, 2017-07-10:
So I was wondering is the use of an opencv Mat the standard way of capturing the ZED stream, or is it also possible from a normal USB path/stream? Does zed - explorer (in the tools) use opencv to display the camera capture?

WASCHMASCHINE, 2017-07-10:
It's an sl::Mat, which usually gets converted to a cv::Mat. And you can even work directly on the GPU with CUDA on it. Unfortunately, there are no samples for the 2.0 SDK. Cannot comment on the Explorer.

santonatos, 2017-07-10:
Yes I am aware of that, I was wondering if there is another way as well. What do you mean directly with CUDA and the GPU? How would that be possible?

WASCHMASCHINE, 2017-07-10:
There is no sample code for 2.0. You could check the archives. There is one GPU OpenGL interop example where you use cudaGraphicsGLRegisterImage to register the CUDA image to a GL_TEXTURE_2D and copy the sl::Mat on the GPU to the CUDA memory each frame grab.

sl-braun, 2017-07-10:
Hi,

ZED Explorer does not use OpenCV for display but OpenGL. It sends the CPU buffer to an OpenGL texture without going through CUDA.

As WASCHMASCHINE just said, you can use the CUDA-OPENGL interop to display a sl::Mat in the most efficient way.

closed 2017-07-13 by santonatos