stereolabs/zed-sdk #15: view retrieved measure

santonatos, 2017-07-14:
So , I am retrieving a disparity measure on a sl:;Mat and I would like to view it. I know that in the depth_sensing sample, you are using an openGL viewer, but because I am getting some errors with that, is there another way?

           Camera zed;
           sl::Mat disparity_measure;
          // Get DISPARITY MEASURE 
          zed.retrieveMeasure(disparity_measure, sl::MEASURE_DISPARITY, sl::MEM_GPU); 
          //view(disparity_measure)??

WASCHMASCHINE, 2017-07-14:
Hey santonatos,
1) You should use sl::MEM_CPU to download the disparity measure to CPU, since you are not using the GPU I assume.
2) I assume you know what a disparity map is. The image consists of a single channel, unlike an RGB image with 3 color channels, and the data are floats. Again in the image, they would be uchars.
3) You can use OpenCV to display float images. There is a function called slMat2cvMat included in the OpenCV example of StereoLabs. You can get a cv::Mat from it and can display it with cv::imshow. You will need to scale the image values between 0 and 1, although. So divide it by 50 or something.

santonatos, 2017-07-14:
hi waschmachine, I am aware of what you are describing, but I am not sure if you can display a disparity mat this way, i think it does not work with a depth map for sure (the depth map uses an openGL viewer, the depth image uses what you are describing)
I will try it, but I am not sure if it will work

WASCHMASCHINE, 2017-07-15:
I can confirm it is working flawlessly with OpenCV and the sl2Mat2cvMat function for the depth measure. I've scaled it by 1/20 for visualization purposes, because maximum depth range is 20m. It works even for float32 images.

santonatos, 2017-07-15:
ok i ll try it. but i care about the disparity not the depth. but the description has them with similar characteristics so that shouldnt be an issue

sl-braun, 2017-07-17:
Hi,
You can do the same with disparity but you need to take care in the "scaling" that disparity is a negative value.
/*OB*/

santonatos, 2017-07-17:
Do you both mean that I would have to go through all the values in the Mat and 'scale them'? How would I scale the values?

WASCHMASCHINE, 2017-07-17:
Come on. Google it. 
cv::Mat newImg = oldDepthImg * 0.05f;

santonatos, 2017-07-17:
lol just that? interesting, This must be a property of the Mat that I am not aware of

sl-braun, 2017-07-17:
Property of cv::Mat, not sl::Mat ( we did not implement it yet).
Therefore just "cast" a sl::Mat into a cv::Mat and then use all the properties of a cv::Mat.

closed 2021-12-01 by bot-stereolabs