stereolabs/zed-sdk #16: change mode to fill

pratikborhade1, 2017-07-17:
I am trying to get the depth map video in Fill mode. As explained in the documentation, I use 
: `sl::RuntimeParameters runtime_parameters;
runtime_parameters.sensing_mode = sl::SENSING_MODE::SENSING_MODE_FILL;`

but it's not really working.
What am I missing here ?

sl-braun, 2017-07-17:
Do you set the runtime_parameters before calling grab() ? 
Maybe post a little bit more code here to see why it is not working

pratikborhade1, 2017-07-17:
yes, I set them underneath initParameters. I am using the sample code itself. The one from the folder SVO recording. I am converting video with depth map enabled.

sl-braun, 2017-07-17:
Therefore I assume you are calling `zed.grab(runtime_parameters); ` on ln84 ? Since grab() without arguments takes default parameters.

pratikborhade1, 2017-07-17:
This is what I was missing. I hadn't passed the parameters in the arguments. Thanks a lot! This solved the problem :)

closed 2017-07-17 by pratikborhade1