stereolabs/zed-sdk #18: view and save depth map after retrieveMeasure

pratikborhade1, 2017-07-27:
I want to view the depth map, so referring to a previous issue I wrote this:

`
sl::Mat depthIm;
zed->retrieveMeasure(depthIm, sl::MEASURE_DEPTH);
cv::Mat tmp_cv;
tmp_cv = cv::Mat(depthIm.getHeight(), depthIm.getWidth(), CV_32FC1, depthIm.getPtr<sl::uchar1>(sl::MEM_CPU)); 
cv::Mat tmp_cv2 = tmp_cv*0.05f;
cv::imshow("View2", tmp_cv2);
`
is it correct ?

and then in order to save the video, what shall I do ?
`
cv::VideoWriter *video_writer = new cv::VideoWriter(<output_path>, int(CV_FOURCC('M',4','S','2'), (int)zed->getCameraFPS(), cv::Size(zed->getResolution().width, zed->getResolution().height));
video_writer->write(tmp_cv2);
`
am I doing anything wrong above ?

WASCHMASCHINE, 2017-07-28:
Hey,
refer to the OpenCV sample if you have any doubts for the 1st part.
https://github.com/stereolabs/zed-opencv/blob/master/src/main.cpp
I think you code is correct, why don't you test it yourself? Maybe scale the tmp_cv because it has values between 0 and 20+ and it should be between 0 and 1 for cv::imshow.

Second part is not related to ZED, but OpenCV.

adujardin, 2017-07-28:
Hi,
Depending on what you want to do, but purely for visualizing the depth map you should use [`retrieveImage(depthIm, sl::VIEW_DEPTH)`](https://www.stereolabs.com/developers/documentation/API/classsl_1_1Camera.html#ac40f337ccc76cacd3412b93f7f4638e2). It will output a normalized (i.e. scaled) 8-bit 1C depth image. Please note that your `cv::Mat` should be a `CV_8UC1` in that case.
Your code seems correct otherwise.

pratikborhade1, 2017-07-28:
@WASCHMASCHINE @adujardin thanks for your replies. Is there a way to save these images in a video or images ? well as it seems the videoWriter doesnt support 32 bits in a single channel and when I saved images with openEXR format, they were only 640bits in size which doesn't make sense. Also, I couldn't open the images with the tool IrfanView which supports openEXR format. @adujardin yes I could visualize the data, but that's not what I want. I need the 32bit depth data.

pratikborhade1, 2017-07-28:
well some of the images were saved properly  with size around 3 MB, but they were still 3 channel images and had only 2 values, either 255 in all three channels or 0 in all three channels. Mostly 255. But retrieveImage() shows me a decent visualization of the depth map. I'm not sure what to do :(

bpinaya, 2017-09-15:
Hi there, any idea why the error: `error: no matching function for call to ‘sl::Camera::retrieveMeasure` is failing? This is for the deep sensing example.
`/home/bpinaya/Documents/gitrepos/zed-examples/depth sensing/src/main.cpp: In function ‘void run()’:
/home/bpinaya/Documents/gitrepos/zed-examples/depth sensing/src/main.cpp:108:85: error: no matching function for call to ‘sl::Camera::retrieveMeasure(sl::Mat&, sl::MEASURE, sl::MEM, int&, int&)’
             zed.retrieveMeasure(point_cloud, MEASURE_XYZRGBA, MEM_GPU, width, height);
                                                                                     ^
In file included from /home/bpinaya/Documents/gitrepos/zed-examples/depth sensing/src/main.cpp:32:0:
/usr/local/zed/include/sl/Camera.hpp:641:20: note: candidate: sl::ERROR_CODE sl::Camera::retrieveMeasure(sl::Mat&, sl::MEASURE, sl::MEM)
         ERROR_CODE retrieveMeasure(Mat &mat, MEASURE measure = MEASURE_DEPTH, MEM type = MEM_CPU);
                    ^
/usr/local/zed/include/sl/Camera.hpp:641:20: note:   candidate expects 3 arguments, 5 provided
CMakeFiles/ZED_Depth_Sensing.dir/build.make:86: recipe for target 'CMakeFiles/ZED_Depth_Sensing.dir/src/main.o' failed
make[2]: *** [CMakeFiles/ZED_Depth_Sensing.dir/src/main.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_Depth_Sensing.dir/all' failed
make[1]: *** [CMakeFiles/ZED_Depth_Sensing.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

adujardin, 2017-09-15:
Hi @bpinaya,
You should update your ZED SDK to [2.1](https://www.stereolabs.com/developers/release/2.1/)
You're probably using 2.0 and the size option with the retrieve functions was introduced later.

closed 2021-12-01 by bot-stereolabs