stereolabs/zed-sdk #19: make error with depth example

nikky4D, 2017-08-09:
Hi,

I am getting this error when compiling with the new samples:

`/usr/local/zed/sample/zed-examples/depth sensing/build$ make 
Scanning dependencies of target ZED_Depth_Sensing
[ 33%] Building CXX object CMakeFiles/ZED_Depth_Sensing.dir/src/main.o
[ 66%] Building CXX object CMakeFiles/ZED_Depth_Sensing.dir/src/GLViewer.o
make[2]: *** No rule to make target '/usr/local/zed/lib/libcudpp.so', needed by 'ZED_Depth_Sensing'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_Depth_Sensing.dir/all' failed
make[1]: *** [CMakeFiles/ZED_Depth_Sensing.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`
Can anyone help me figure out what is wrong?

WASCHMASCHINE, 2017-08-09:
libcudpp sounds like CUDA. Did you install it?

nikky4D, 2017-08-09:
Thanks for the fast response. I just solved the problem. I removed my zed folder from /usr/local, and reinstalled it again with the .run file. That appears to have fixed the error and everything compiles and runs well. 

I use Cuda-8 and it runs well for everything else.

closed 2017-08-25 by nikky4D