stereolabs/zed-sdk #20: drop in frames in real time mode (retrieveDepth())

pratikborhade1, 2017-08-10:
Hey,
so I am calling the retriveMeasure() after the zed Grab() function to get the depth from the camera in the real time mode. But so many frames are being dropped without being processed. For example around 70 million frames were dropped when only 2000 frames were processed. I tried with svo_real_time_mode = true/false both. But got similar results. Is there a way I can fix this problem ?

adujardin, 2017-08-10:
Hi,
`svo_real_time_mode` works only is SVO playback and not in live mode. Also a 70 million frames SVO seems absolutely huge and must take terabytes (in 720p@60 that's around 2 weeks of continuous video). There might be an issue with the SVO (frame count or timestamp). Which compression did you use for the SVO recording ? The RAW format doesn't handle the timestamp so the real time mode is not compatible with those either. 
Could you retry launching your program in verbose mode ? What's your configuration (ZED SDK version, OS, GPU) ?

pratikborhade1, 2017-08-10:
I'm sorry, I didn't know how to put it in words so I used the word frames dropped. But actually I had separate counters everytime zed.grab() was a success (count1) and when it wasn't (count2). So lets say when count1 was 2000, count 2 was around 70 Million. Maybe this will help. When I used a video to get depth with retrieveMeasure(), 20-21 frames were processed per second. Whereas, only 4.6 were processed in live mode. . I used standard clock to find out the time for processing. Is there a way to better this performance ? Could it be an OS problem ?

Configuration:
ZED SDK 2.1
Windows 10
i7 -4702HQ
16gb ram
Quadro K1100M GPU

Thanks!

adujardin, 2017-08-10:
Oh ok, I see. The success status of `grab()` can be kind a misleading. It means that a new image was found and the computation went well. In your case when it "fails" it probably just means that there isn't a new frame yet, it needs to sleep for a few ms. You should check the [value of the error code](https://www.stereolabs.com/developers/documentation/API/group__Enumerations.html#details), it's probably `ERROR_CODE_NOT_A_NEW_FRAME`.
You should measure the time/count the frame only when the grab return `SUCCESS`.

pratikborhade1, 2017-08-10:
Yes, you are right. I get it now. Can you recommend the best possible way to get the depth map(retrieveImage()) at minimum 30fps while using ZED camera in live mode? I even have a more powerful system.

closed 2021-12-01 by bot-stereolabs