stereolabs/zed-sdk #21: SVO_Recording does convert .svo to .avi or .png

RSKothari, 2017-08-18:
For some reason, it doesn't convert an input .svo video to a .avi video or .png image files. It writes out a few .png images but they are all black.

Is it because the ZED SDK wasn't compiled correctly?

WASCHMASCHINE, 2017-08-19:
Have you inspected the images? Are they pitch black, like 0?

RSKothari, 2017-08-20:
Hey @WASCHMASCHINE , 
Yes, the images are pitch black, as in 0. For some reason, The code runs smoothly and outputs nothing.

RSKothari, 2017-08-29:
@WASCHMASCHINE Bump.

WASCHMASCHINE, 2017-08-30:
Post your code and add more info: Which SDK version etc.

RSKothari, 2017-08-30:
I found the fix to the solution. I used cmake-gui to look at the variables and the by default setting for OpenCV_DIR was pointed to the cmake folder in my opencv folder. I changed it to the build directory and manually set the flag OpenCV_FOUND to 1. Export and Playback work alright now. Thanks @WASCHMASCHINE !

closed 2017-08-30 by RSKothari