stereolabs/zed-sdk #22: Multi-Camera (6 needed)

Choonghee-frank, 2017-08-19:
Hello, I am trying to use 6 cameras for take the video recording at the same time (synchronized).

Since it is only available for using 4 cameras for the same time, are there any ways to synchronize all 6 cameras?
(Suppose we put 2 laptops and 3 ZED cameras for each laptop, how can we synchronize those video recording ?)

WASCHMASCHINE, 2017-08-19:
Do you use ROS? This would allow 4 cameras per PC connected in a network because you could customize the topic names.

I assume ROS offers some way of synching the timestamp of your frames via the network. So this not necessarily a ZED issue, but a ROS one.

mitjaprelovsek, 2017-10-23:
Hi,

Just curious: you are talking about Ubuntu, right? There is no way of using multiple ZEDs on a Win64 platform as of yet, is there?

Thanks!

WASCHMASCHINE, 2017-10-23:
No way of Win64 capture as of now. Still, I've played back two cameras on Windows via Rosbags.

AKANK16, 2017-11-20:
I need to use the video feeds of 6 Zed Cameras.These feeds are required to be used at the same time on Jetson TX2.How do i possibly acheive it?

qiao19881314, 2017-12-05:
@Choonghee-frank    Hi  how you  recoding the  the videos of  4 cameras at the same time?
using the ROS or some other methods?

WASCHMASCHINE, 2017-12-05:
@qiao19881314 It is possible using ROS and Ubuntu. I tried it with 2 cameras at the moment. Make sure your pc handle it (separate usb3 controllers).

qiao19881314, 2017-12-05:
@WASCHMASCHINE thank u. Can we recording the svo files of the two cameras synchronously？Could u share this code with me?

WASCHMASCHINE, 2017-12-05:
@qiao19881314 I don't think it's possible.

WASCHMASCHINE, 2017-12-05:
But you can record Rosbags, which is similar.

Choonghee-frank, 2017-12-05:
@qiao19881314 
Oh, I forgot about this thread. You can record the camera synchronously(in svo files), since I modified the code, you can also find the code by searching 'multi-zed camera'

Choonghee-frank, 2017-12-05:
@WASCHMASCHINE 
Thanks for your reply, I've done with 2 cameras and also with other experimental devices such as encoder and force plate. I was also able to connect 2 cameras in a single laptop. However, do you have any idea how to connect multiple cameras?

WASCHMASCHINE, 2017-12-06:
What do you mean with multiple cameras? You already have two connected, which is "multiple".
If you have a quad bus host controller for the USB3 then you could connect four at a time.

USB3 is enought for one camera, so sending the data of say four zed cameras through the same channel would not work that well. You will lose a lot of frames, even on low res.

qiao19881314, 2017-12-10:
@Choonghee-frank 
  Thank you. Do you means this code https://github.com/adujardin/zed-multi-input?
  Since it using a for loop,  the two SVO files are not time synchronized.  For here, i want that the  each  frame in the two svo files are synchronously. Do you have ides for that?

qiao19881314, 2017-12-10:
@WASCHMASCHINE    Thank you. I already realize the recording using the rosbag.  However time synchronized still a problems.  The messages recording in ROS seems also is serial recording.   When i recording the left  right images  and 3d point cloud from  two zed cameras, they are not same numbers. Could you have good ideas for time synchronization?

WASCHMASCHINE, 2017-12-11:
@qiao19881314 Are you using the ROS sample for multi-Input?
For the images, you could try if there is still a side-by-side option. So you only have one image.
You could also try to reorder the images according to the frame id at the receiving end.

jm677zz, 2018-01-09:
Six Synchronized Full HD Cameras for NVIDIA® Jetson TX1/TX2: The e-CAM30_HEXCUTX2 streams 720p(HD), 1080p (Full HD) and 3.4MP @ 30fps in uncompressed YUV422 Format in both synchronous and Asynchronous mode.

https://www.e-consystems.com/multiple-csi-cameras-for-nvidia-jetson-tx2.asp

Choonghee-frank, 2018-01-29:
@WASCHMASCHINE  Sorry for the late response. In our system, yes we use the camera with for loops and we tried to compare the timestamp between each cameras.
The time gap between the two cameras was small enough(less than 1 ms) since we are looking for a human motion. I don't know the system you are trying to analyze, but I think using the for-loop is the easiest way to analyze it in synchronized way. I don't know how close you want to make the system synchronized (in terms of time gaps?)

WASCHMASCHINE, 2018-01-29:
@Choonghee-frank What exactly do you mean with a for loop? You basically grab camera 1 and then camera 2?

closed 2021-12-01 by bot-stereolabs