stereolabs/zed-sdk #24: std::string to sl::String?

antithing, 2017-08-31:
Hi guys, i need to concatenate a std::string with an sl::String. How can i do this?

Thanks!

WASCHMASCHINE, 2017-09-01:
https://www.stereolabs.com/developers/documentation/API/classsl_1_1String.html

One way is to get a C string via sl::String::c_str(). The std variant has the same function I think. 
Then your problem is about concatenating C strings. You can use something like strcat for that:
https://stackoverflow.com/questions/308695/how-do-i-concatenate-const-literal-strings-in-c

antithing, 2017-09-01:
Hi, thanks for your reply. My issue is I need to pass a std::string filepath to  the `savePointCloudAs` function.

So I have:

std::string dir =  "C/Temp";
sl::String name = "pointCloud";

How can i pass the combined path to `savePointCloudAs` ?

thanks!

WASCHMASCHINE, 2017-09-01:
sl::String offers a constructor for a C string. (If you do it the way I described before)

Else, you might also try casting the sl one to std, combine them as an std via + and then cast them back to sl.

I am not trying to be rude, but you may need to brush up on some C++.

antithing, 2017-09-01:
Thanks! It is the casting back to sl that I am confused about. Say That i have a `std::string tmp`, how would I cast that to an sl::String?

closed 2017-09-02 by antithing