stereolabs/zed-sdk #27: error: ‘slMat2cvMat’ was not declared in this scope

TheMikeyR, 2017-09-27:
I'm trying to build the svo_recording/export and playback, but I get the following  `error: ‘slMat2cvMat’ was not declared in this scope`
The other samples works just fine.

```
(cv)  /usr/local/zed/sample/svo recording/export  mkdir build && cd build && cmake .. && make
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable exact version "8.0") 
-- Found OpenCV: /usr/local (found version "3.2.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/zed/sample/svo recording/export/build
Scanning dependencies of target ZED_SVO_Export
[ 50%] Building CXX object CMakeFiles/ZED_SVO_Export.dir/src/main.o
/usr/local/zed/sample/svo recording/export/src/main.cpp: In function ‘int main(int, char**)’:
/usr/local/zed/sample/svo recording/export/src/main.cpp:83:52: error: ‘slMat2cvMat’ was not declared in this scope
     cv::Mat left_image_ocv = slMat2cvMat(left_image);
                                                    ^
CMakeFiles/ZED_SVO_Export.dir/build.make:62: recipe for target 'CMakeFiles/ZED_SVO_Export.dir/src/main.o' failed
make[2]: *** [CMakeFiles/ZED_SVO_Export.dir/src/main.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_SVO_Export.dir/all' failed
make[1]: *** [CMakeFiles/ZED_SVO_Export.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

```
(cv)  /usr/local/zed/sample/svo recording/playback  mkdir build && cd build && cmake .. && make
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable exact version "8.0") 
-- Found OpenCV: /usr/local (found version "3.2.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/zed/sample/svo recording/playback/build
Scanning dependencies of target ZED_SVO_Playback
[ 50%] Building CXX object CMakeFiles/ZED_SVO_Playback.dir/src/main.o
/usr/local/zed/sample/svo recording/playback/src/main.cpp: In function ‘int main(int, char**)’:
/usr/local/zed/sample/svo recording/playback/src/main.cpp:64:50: error: ‘slMat2cvMat’ was not declared in this scope
     cv::Mat svo_image_ocv = slMat2cvMat(svo_image);
                                                  ^
CMakeFiles/ZED_SVO_Playback.dir/build.make:62: recipe for target 'CMakeFiles/ZED_SVO_Playback.dir/src/main.o' failed
make[2]: *** [CMakeFiles/ZED_SVO_Playback.dir/src/main.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_SVO_Playback.dir/all' failed
make[1]: *** [CMakeFiles/ZED_SVO_Playback.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

WASCHMASCHINE, 2017-09-28:
Hi @TheMikeyR,
include the slMat2cvMat function from https://github.com/stereolabs/zed-opencv/blob/master/src/main.cpp (OpenCV example) in the export main.cpp.

TheMikeyR, 2017-10-26:
@WASCHMASCHINE sorry for late answer, had to use the export now and your comment fixed my issue, I've did this to get it to work:
Added this line after namespaces https://github.com/stereolabs/zed-opencv/blob/abc6db4b0b7678fce0cbd1d53e93704f5b860d8e/src/main.cpp#L31

Added this function to the bottom of the file
https://github.com/stereolabs/zed-opencv/blob/abc6db4b0b7678fce0cbd1d53e93704f5b860d8e/src/main.cpp#L103

Then it compiles succesfully and I'm able to export .svo files to .avi files

closed 2017-10-26 by TheMikeyR

WASCHMASCHINE, 2017-10-26:
I think you could have also included "utils.hpp" from the SDK SVO playback example, @TheMikeyR

TheMikeyR, 2017-11-01:
@WASCHMASCHINE Yeah you are right, in the end I pulled the sample from github and build it since the version in the ZED SDK was lacking some export options.