stereolabs/zed-sdk #29: Can ZED do SLAM?

callmeray, 2017-10-22:
Can ZED do SLAM?  If so, what algorithm is used? Is IMU used?
Thank you!

WASCHMASCHINE, 2017-10-22:
1. Yes, with the ZedFu app. Or do your own.
2. Undisclosed. Maybe an ZedFu is an adaption of PCL's KinFu, who knows (Stereolabs does).
3. No, the ZED has no IMU.

callmeray, 2017-10-23:
@WASCHMASCHINE Thank you for your answers. One more question, what's the depth accuracy of ZED?

WASCHMASCHINE, 2017-10-23:
@callmeray To be honest, I just returned to my PC after taking some measurements (~50) for that purpose. I used a laser for point distance attached to the ZED and moved a textured target away from the setup from 1 to 25 m. Now it says 12m in the Depth Viewer (which is even the distance!) while the laser says 25.137m. 

I've done the experiment before, but I did not record SVOs. The last result was that I could add an exponential offset to the distance to fit my laser measurements, which did help. I contacted Stereolabs with the data, but they said they will add depth stabilization in the next update, which did not help apparently.

Some scenarios:
a) My camera may be broken
b) The factory calibration is off
c) Their stereo algorithm is wrong or at least the disparity to depth step
d) They do not account for their own calibration value, i.e. the convergent factor. So the camera may be a little cross-eyed, and you may need to add an offset to the disparity, which results in nonlinear depth offset (like observed).
e) The camera could not work under the lighting

Anyway, I have to analyze the SVOs first. Still, the ZED is a solid stereo camera, even though the software might not be delivering me the right results right now. You can still get the raw data on the GPU, which is great.

RSKothari, 2017-10-23:
@WASCHMASCHINE Does it mean that the Depth Viewer is off by 12m for a ground truth depth of 25m? Theoretically, if you can map out a grid in depth then you could use the ground truth values to create a 3D interpolation function (splines or a non-linear fitting would be best IMO). Offsets to the depth values can be corrected offline using this approach.

WASCHMASCHINE, 2017-10-23:
@RSKothari Yes, that's what I meant. But only for a point in roughly the image center. I did find a non-linear fit for the depth in the first experiment and used it for every depth. But now I want to find the source of the problem, not fighting the symptoms with an adhoc calibration, again.

callmeray, 2017-10-24:
@WASCHMASCHINE  Thank you so much for your work. According to what you said, I think the most probable reason is that  the factory calibration is off and the system needs to be recalibrated

closed 2018-01-24 by callmeray