stereolabs/zed-sdk #34: Ubuntu 17.10 ZED Detected, but cannot open Camera.

Sijoma, 2017-10-31:
Hey guys,

I know that 17.10 is still an unsupported release but I think everything works. (dependencies etc.) .
However i'm not able to open the camera.

I get this error. 

*** Download Factory Calibration ***
ZED Detected, but cannot open Camera. Please check that you don't have another camera connected...
ZED SDK installation complete !

Is there some trick i am missing? Or do I have to wait till Ubuntu 17.10 is supported? I'm running kernel 4.13.0-16-generic

Sijoma, 2017-11-13:
I just installed Ubuntu 16.04 LTS and I get the same problem.

adujardin, 2017-11-14:
Hi,
You should launch the diagnostic tool (`/usr/local/zed/tools/ZED\ Diagnostic`). The ZED SDK should work with Ubuntu 16.04 (with the current kernel at least, which is 4.4). 
It is possible that a newer kernel is not handled. We'll also look into it.

Sijoma, 2017-11-14:
This is the output of the tool. I am running kernel 4.13. I will try later with older kernels.

ZED Diagnostic Tool

Running ZED SDK Components Diagnostic : OK  

Running hardware Diagnostic : OK  

Running cuda Diagnostic : OK  

Running usb Diagnostic : OK  

Segmentation fault (core dumped)

adujardin, 2017-11-14:
I found the issues, V4L device descriptor changed a little with the newer kernel, the checks were not expected the new output value. The fix should be included in the next release.

Sijoma, 2017-11-14:
Is there an ETA for that release?

adujardin, 2017-11-14:
The release is planned for December.

adujardin, 2017-11-19:
Hi,
Just to let you know that we decided to include the patch in a bugfix release: [ZED SDK 2.2.1 ](https://stereolabs.com/developers/release/2.2/#sdkdownloads_anchor)

Sijoma, 2017-11-19:
Thanks for the information! I will try that out!

Sijoma, 2017-11-22:
Hey, 

the update worked for my laptop running the higher kernel. However i am having problems getting the ZED running on the Nvidia Jetson TX2.  Note: The Jetson is running Kernel 4.4

The camera is detected with 'lsusb' and the error of the ZED Diagnostic Tool is the same:

ZED Diagnostic Tool

Running ZED SDK Components Diagnostic : OK

Running hardware Diagnostic : OK

Running cuda Diagnostic : OK

Running usb Diagnostic : OK

Segmentation fault (core dumped)

adujardin, 2017-11-23:
I can't reproduce the issue, it should work. 
Could you contact the support (support@stereolabs.com) to check what's wrong?

Sijoma, 2017-11-23:
I will do that! Thanks

marnix93, 2018-01-15:
Hi, I have exactly the same problem. When installing the SDK I get the message: "ZED Detected, but cannot open Camera. Please check that you don't have another camera connected..." When running ZED Diagnostic everything is fine but a segmentation fault pops up. My SDK version is 2.2.1 and my Linux kernel is 4.13. A few weeks ago, everything worked fine.. Can anyone give a suggestion for me?

RSKothari, 2018-01-16:
@marnix93 What's your CUDA version?
@adujardin Does the update patch version 2.2.1 work with CUDA 8.0?

marnix93, 2018-01-16:
I am using CUDA 8.0

RSKothari, 2018-01-16:
[image]

The update patch might not work with CUDA 8.0. The website doesn't list it on there.

adujardin, 2018-01-16:
The ZED SDK 2.2.1 doesn't work with CUDA 8.

@marnix93 are you sure you're not using the ZED SDK 2.2.0 (which doesn't include the fix for the new kernel)?

marnix93, 2018-01-16:
Ok, now I see. I did not look carefully to the version names and just thought that the SDK listed under CUDA 8 was the newest one. Upgrading to CUDA 9 is at this moment too complicated thus therefore I will stick to the older kernel. Thanks for the comments.

RSKothari, 2018-01-16:
@marnix93 It's a whole lot easier to simply use an earlier Ubuntu kernel. You can choose one during GRUB boot loading. If you want to permanently change which kernel to load from:
https://unix.stackexchange.com/questions/198003/set-default-kernel-in-grub

marnix93, 2018-01-17:
@RSKothari This is exactly what I did ;). I only did it in a less elegant way by removing the Linux Kernel 4.13