stereolabs/zed-sdk #37: std::length_error while saving mesh in spatial mapping example using vs2017.5

seraphain, 2017-12-12:
**Hardware**
Intel Core i7-4810MQ
16G RAM
GeForce GTX 765M 2GB

**Software**
Windows 10 1709
Visual Studio 2017.5
CUDA 9.0.176
CMAKE 3.10.0
PCL 1.8.1

When executing mesh.save("mesh.obj"), the application crashed and throwed std::length_error.
Debug and find that _explicit vector(_CRT_GUARDOVERFLOW const size_type _Count, const _Alloc& _Al = _Alloc())_ was called with _Count equal to 14757395255531667456 first then 0, and the application crashed.

wysk0101, 2018-02-06:
The same "std::length_error "as I faced,    WeChat:   wysk0101

AkashPushkar, 2018-04-26:
Did anyone found a solution for this?

rkachach, 2018-06-01:
Same error here and the same issue is causing a crash in several other examples.

nesnes, 2018-06-01:
Hi, can you make sure that you're running in Release mode? Th issue looks like compatibility issues in the `std::vector` implementation in debug mode (your app) and release (the ZED SDK).

rkachach, 2018-06-01:
Thanks for the quick and concise reply. I have just switch to Release mode and the issue doesn't occur any more.  👍

closed 2018-06-01 by nesnes

Wanggcong, 2022-10-27:
> Thanks for the quick and concise reply. I have just switch to Release mode and the issue doesn't occur any more. +1

Hi, I have similar errors. Does switching to Release mode mean change the git branch to release_3.6? Thank you!