stereolabs/zed-sdk #41: Where sl_zed / Camera.hpp is located?

Declining, 2018-01-21:
Hey ,guys. 
When I tried to compile the code in tutorial, the head file "sl_zed/Camera.hpp" cannot be found. 
So I want to know where I can find this header file, or do I need to create another environment?

adujardin, 2018-01-21:
Hi,
Replace "sl_zed" by "sl" and it should work (`#include "sl/Camera.hpp"`). 

It was originally meant for the new SDK (2.3) but it's not mandatory anymore. I will fix this in the repository.

closed 2018-01-24 by adujardin

Declining, 2018-01-24:
Thank you, I have solved this problem by you help.

Closed #41<https://github.com/stereolabs/zed-examples/issues/41> via 398893d<https://github.com/stereolabs/zed-examples/commit/398893d538e35324a19076543947383dfe9b82d8>.

schliffen, 2019-05-24:
It was not fixed yet, I'm still facing with that error!

obraun-sl, 2019-05-24:
Make sure that the ZED SDK include directory is in the include directories list. If using Cmake , it should be automatically done.
For example, under windows, the include dir is C:/Program Files(x86)/ZED SDK/include

Vizalkar, 2020-02-28:
Hi
I encountered the same error and i use cmake.
It is not solvec, the directory in ZED SDK is "sl", not "sl_zed".
Changing the #include "sl_zed/*" lines does the trick

commit 398893d5 (2018-01-24): Fix #41 headers path 2.3