stereolabs/zed-sdk #42: SVO Recording very CPU intensive

cosama, 2018-02-09:
When running the recording example (as it comes) on a Jetson TX2 with a ZED Mini it uses almost 3 CPU cores at 100% each. It seems that the problem is the compression, as on SVO_COMPRESSION_MODE_RAW, the CPU usage is only at about 1 CPU core at 100% usage.

To improve the situation I tried to drop the framer rate by adding  `'initParameters.camera_fps  = 30;`. It shows that it records only at 30fps, however there is no improvement what so ever in the CPU usage, so I assume that setting is ignored in recording mode.

Our application requires low power, and as we can not run full mapping/tracking for the entire time, we were hoping that it would be possible to store (buffer) the stream to a file (or even better keep it in RAM) and then replay it when we need it.

Is there a way to get about that which doesn't drain the resources as much? What is the situation with the RAW mode? It says it is deprecated and doesn't support IMU data. Is this true?

Lejboelle, 2018-10-08:
Bump,
Also interested in lowering the CPU usage, as I'm trying to record from a ZED camera and Realsense camera simultaneously, but getting lots of frame drops for the Realsense stream do to CPU usage by ZED. Already tested that issue is not bandwidth.

cosama, 2018-10-08:
The bottle neck here seems to be the compression, so you could just dump the images in a binary file yourself after grabbing them. However, if you require playback through the SDK I think it is necessary to store images at the highest rate possible anyways. Lowering the resolution should help a bit.

I have the impression Stereolab has been working quite intensely on lowering the consumption of tracking, so you might just want to run tracking in real time and store the tracking result together with your images for further processing. At least on our system running tracking and store images as binary files is much less resource intensive than svo recording.

obraun-sl, 2018-10-08:
Hi,
On the upcoming version (2.7.0), we will introduce a new SVO compression mode, based on NVENC compression (AVCHD or HEVC) . This will allow a huge reduction of size of the SVO files and almost no CPU/GPU usage (since it' using a dedicated hardware).
You will need a NVIDIA GPU to have this mode available but I think that would help your case.

Best,
OB

ftheirs, 2019-09-30:
Hi!
I tried SVO recording example with H264 and H265 encoding in my Jetson TX2 but the CPU usage is between 100 and 120%. I have installed ZED SDK 2.8.2. Is there any update on this issue?
Frame rate: 30fps
Resolution: HD720
Compression: SVO_COMPRESSION_MODE_HEVC / SVO_COMPRESSION_MODE_AVCHD

How can I check that I am using hardware encoding on Jetson platform?
[image]

closed 2021-12-01 by bot-stereolabs