stereolabs/zed-sdk #48: Jetson TX2 + Jetpack 3.2 + Orbitty carrier board

Kruisheer, 2018-03-09:
I did not manage to run the Zed Stereo camera on jetson TX2 with Orbitty using the Jetpack 3.2. I thought it was a bad cable issue at first but now I feel it could be a software issue.  It worked using Jetpack 3.1 and works well on my laptop. I am wondering if any knows what the issue could be. 

When Running the diagnostic tool it shows that the USB bandwidth is too low
Frame drop: 57/480
Frame tearing: 3/480
Camera: OK
[ZED_Diagnostic_Results.txt](https://github.com/stereolabs/zed-examples/files/1796670/ZED_Diagnostic_Results.txt)

adujardin, 2018-03-09:
Hi,
You should get in touch with the support (support@stererolabs.com) for this issue.

ravikt, 2018-06-27:
Hi, I ran the ZED stereo cam on the Orbitty TX2 board. However, I get the USB error when I run the diagnostic tool. I am using a powered USB3.0 hub for the camera. Other tools like explorer, depth viewer seems to work fine. But the ZedFu shows an error "Zed seems to be running on USB2. cannot activate mapping".  Can anyone suggest some solution ?

closed 2021-12-01 by bot-stereolabs