stereolabs/zed-sdk #51: Can't convert svo to avi video

ThomasToys, 2018-03-28:
Hi,
I'm trying to convert a recorded test video to an avi file using the resources in this repository.
I'm stuck on the converting step :
Zed_SVO_Export "C:\Users\Username\Videos\test_1080p_60fps.svo" "C:\Users\Username\Videos\result.avi" 0

Can't figure out why i got the following error : 
OpenCV video writer cannot be opened. Please check the .avi file path and write permissions.

It seems the video writer is not properly opened at line 135 of main.cpp script:
video_writer = new cv::VideoWriter(output_path, fourcc, frame_rate, image_size_sbs);

Any help is appreciated!
Thanks in advance.

B-Betty, 2018-04-12:
hi. I also get the same question. do you solve the problem??

ThomasToys, 2018-04-23:
Might be a codec issue. I make the image sequence conversion work. I did not manage to find out how to solve the avi conversion though.
I'll post the solution if I find it.

rishabh700, 2019-07-01:
Are you running the Command Prompt as Admin?

chingchichuang, 2019-10-08:
Hi,
I have the same issue and may have the answer.
Thanks for the hint from ThomasToys (i.e. codec issue).
I copy my opencv_ffmpeg3**_64.dll from C:\Program Files (x86)\ZED SDK\dependencies\opencv_3.4.1\build\x64\vc14\bin to \Path\to\zed_export\build\Release
By doing this, my problem was solved.

closed 2021-12-01 by bot-stereolabs

JaviBaeza, 2022-02-15:
When you download the last sdk version it happens again. Any idea why?

krizzagit, 2022-08-15:
Also looking for an answer, this is the following error I get: 

(python:12245): GStreamer-CRITICAL **: 11:18:24.286: gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed
[ WARN:0@0.599] global /opt/conda/conda-bld/opencv-suite_1659973592936/work/modules/videoio/src/cap_gstreamer.cpp (1813) close_ OpenCV | GStreamer warning: No source in GStreamer pipeline. Ignore
[ERROR:0@0.599] global /opt/conda/conda-bld/opencv-suite_1659973592936/work/modules/videoio/src/cap.cpp (595) open VIDEOIO(GSTREAMER): raised OpenCV exception:

OpenCV(4.5.5) /opt/conda/conda-bld/opencv-suite_1659973592936/work/modules/videoio/src/cap_gstreamer.cpp:147: error: (-215:Assertion failed) ptr in function 'get'

[ERROR:0@0.600] global /opt/conda/conda-bld/opencv-suite_1659973592936/work/modules/videoio/src/cap.cpp (595) open VIDEOIO(CV_IMAGES): raised OpenCV exception:

OpenCV(4.5.5) /opt/conda/conda-bld/opencv-suite_1659973592936/work/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): ~/Documents/ZED2_Video_Export/output_file.avi in function 'icvExtractPattern'

OpenCV video writer cannot be opened. Please check the .avi file path and write permissions.

SzaremehrjardiMT, 2022-11-28:
I added /output.avi (it should be made by Zed_SVO_Export) at the end of my output folder, and it fixed the issue.

jjw1230, 2024-03-02:
Usage:

    ZED_SVO_Export A B C

Please use the following parameters from the command line:
 A - SVO file path (input) : "path/to/file.svo"
 B - AVI file path (output) or image sequence folder(output) : "path/to/output/file.avi" or "path/to/output/folder"
 C - Export mode:  0=Export LEFT+RIGHT AVI.
                   1=Export LEFT+DEPTH_VIEW AVI.
                   2=Export LEFT+RIGHT image sequence.
                   3=Export LEFT+DEPTH_VIEW image sequence.
                   4=Export LEFT+DEPTH_16Bit image sequence.
 A and B need to end with '/' or '\'

Examples:
  (AVI LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 0
  (AVI LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 1
  (SEQUENCE LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 2
  (SEQUENCE LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 3
  (SEQUENCE LEFT+DEPTH_16Bit)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 4

Press [Enter] to continue

jjw1230, 2024-03-02:
Usage:

    ZED_SVO_Export A B C

Please use the following parameters from the command line:
 A - SVO file path (input) : "path/to/file.svo"
 B - AVI file path (output) or image sequence folder(output) : "path/to/output/file.avi" or "path/to/output/folder"
 C - Export mode:  0=Export LEFT+RIGHT AVI.
                   1=Export LEFT+DEPTH_VIEW AVI.
                   2=Export LEFT+RIGHT image sequence.
                   3=Export LEFT+DEPTH_VIEW image sequence.
                   4=Export LEFT+DEPTH_16Bit image sequence.
 A and B need to end with '/' or '\'

Examples:
  (AVI LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 0
  (AVI LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 1
  (SEQUENCE LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 2
  (SEQUENCE LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 3
  (SEQUENCE LEFT+DEPTH_16Bit)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 4

Check this carefully you will find out that the  "path/to/output/file.avi" there is an ".avi" we may forget to it

jjw1230, 2024-03-02:
Usage:

    ZED_SVO_Export A B C

Please use the following parameters from the command line:
 A - SVO file path (input) : "path/to/file.svo"
 B - AVI file path (output) or image sequence folder(output) : "path/to/output/file.avi" or "path/to/output/folder"
 C - Export mode:  0=Export LEFT+RIGHT AVI.
                   1=Export LEFT+DEPTH_VIEW AVI.
                   2=Export LEFT+RIGHT image sequence.
                   3=Export LEFT+DEPTH_VIEW image sequence.
                   4=Export LEFT+DEPTH_16Bit image sequence.
 A and B need to end with '/' or '\'

Examples:
  (AVI LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "**path/to/output/file.avi**" 0
  (AVI LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 1
  (SEQUENCE LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 2
  (SEQUENCE LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 3
  (SEQUENCE LEFT+DEPTH_16Bit)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 4

**Check this carefully you will find out that the "path/to/output/file.avi" there is an ".avi" we may forget to it**

jjw1230, 2024-03-02:
Usage:

ZED_SVO_Export A B C
Please use the following parameters from the command line:
A - SVO file path (input) : "path/to/file.svo"
B - AVI file path (output) or image sequence folder(output) : "path/to/output/file.avi" or "path/to/output/folder"
C - Export mode: 0=Export LEFT+RIGHT AVI.
1=Export LEFT+DEPTH_VIEW AVI.
2=Export LEFT+RIGHT image sequence.
3=Export LEFT+DEPTH_VIEW image sequence.
4=Export LEFT+DEPTH_16Bit image sequence.
A and B need to end with '/' or ''

Examples:
(AVI LEFT+RIGHT) ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 0
(AVI LEFT+DEPTH) ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 1
(SEQUENCE LEFT+RIGHT) ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 2
(SEQUENCE LEFT+DEPTH) ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 3
(SEQUENCE LEFT+DEPTH_16Bit) ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 4

Check this carefully you will find out that the "path/to/output/file.avi" there is an ".avi" we can not forget to it