stereolabs/zed-sdk #57: Deep copy by constructor doesn't work

lazysquid, 2018-04-24:
Hello. I found some weird problem related to *sl::Mat* class. 
But I couldn't find some places to report this issue so I open it at here.

ZED API doc says that I can deep copy from one matrix to another matrix by using [constructor](https://www.stereolabs.com/developers/documentation/API/v2.3.3/classsl_1_1Mat.html#a9cc42063b353ce75ee279f48d46cd47d)
 of the class.
However, it only conduct shallow copy instead of deep copy.
I did some experiments like in below.

```cpp
sl::Mat mat;
zed.retrieveMeasure(mat, sl::MEASURE_XYZRGBA, sl::MEM_CPU, width, height);
sl::Mat mat2;
mat2.setFrom(mat);
sl::Mat mat3(mat); // I also tested static_cast<const sl::Mat&>(mat)) but it works same.
std::vector<sl::Mat> mats;
mats.push_back(mat);
std::cout << mat.getInfos() << "\n";
std::cout << mat2.getInfos() << "\n";
std::cout << mat3.getInfos() << "\n";
std::cout << mats[0].getInfos() << "\n";
```
The code above spited out messages in below

>C714D270 Ts 1524577055879201960 sl::Mat of size [1280,720], with 4 channels of type float allocated on CPU (memory owned).
C714D2E0 Ts 1524577055879201960 sl::Mat of size [1280,720], with 4 channels of type float allocated on CPU (memory owned).
C714D350 Ts 1524577055879201960 sl::Mat of size [1280,720], with 4 channels of type float allocated on CPU (**reference** only).
7FE07F0 Ts 1524577055879201960 sl::Mat of size [1280,720], with 4 channels of type float allocated on CPU (reference only).

Is there anything that I'm missing? Or is it just ZED SDK problem?

adujardin, 2018-04-24:
Hi,
Thanks for letting us now. 
It is an issue, the API doc is wrong. It is actually meant to be a shallow copy.

lazysquid, 2018-04-24:
Thanks for fast feedback.
Okay.
Should I open this issue until it is fixed or just close it right now? 
Also is there any official place for ZED that I can have nice Q&A such as develop forum?

adujardin, 2018-04-24:
You can close it, I'll push a fix for the next release.
We currently don't have such a place. However, it's planned to have a forum. It should be online soon.

lazysquid, 2018-04-25:
Okay. Then I'll close the issue.

closed 2018-04-25 by lazysquid