stereolabs/zed-sdk #59: ZED not detected in Ubuntu 16.04

Psunshine, 2018-05-02:
I stuck on this problem all the day.... /(ㄒoㄒ)/~~
My laptop has two systems, Ubuntu 16.04 and Windows pro 10. I tested ZED camera in windows 10 and it works normally. But in ubuntu, it says 'ZED not detected'. 
I tried cuda9.0+ZED_SDK_Linux_Ubuntu16_v2.2.1.run and cuda8.0+ZED_SDK_Linux_Ubuntu16_CUDA8_v2.2.0.run, I successed install cuda9.1 but failed to change the GPU device to NVIDIA GPU in `NVIDIA X Server device settings`.

This is the kernel infomation:
`
Linux T470p 4.13.0-39-generic #44~16.04.1-Ubuntu SMP Thu Apr 5 16:43:10 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
`

The output of `ZED Diagnostic`:
`
ZED Diagnostic Tool

Running ZED SDK Components Diagnostic : OK  

Running hardware Diagnostic : OK  

Running cuda Diagnostic : OK  

Running usb Diagnostic : Failed 
Error:
    ZED not detected, check your USB 3.0 drivers and avoid hubs or cable extension.

Running zed Diagnostic : Failed 
Error:
    ZED not detected. Connect it to an USB 3.0 port. (Blue USB ports)

Diagnostic finished, a file has been generated : ZED_Diagnostic_log.json

`

And the contents of `ZED_Diagnostic_log.json`:
`
{
    "ZED SDK Components": {
        "DiagnosticZEDSDKVersion": "2.2.0"
    },
    "cuda": {
        "CUDA Toolkit version": "V8.0.61",
        "SMI": "(SMI) :     Performance State               : P0",
        "deviceCount": 1,
        "deviceDriverVersion": 9000,
        "devices": [
            {
                "arch": "Maxwell",
                "computeCapability": "5.0",
                "cores": 384,
                "name": "GeForce 940MX",
                "totalMemoryBytes": "2100232192",
                "totalMemoryMB": 2002.9375
            }
        ],
        "initResult": 0,
        "loaded": true
    },
    "hardware": {
        "OS": "Ubuntu 16.04.4 LTS ",
        "PCName": "T470p",
        "avxSupported": true,
        "coreCount": "8",
        "cpu": " Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz",
        "datetime": "2018-05-02 17时57分59秒",
        "hyperThreading": "0",
        "ramCapacity": "16.552747",
        "ramUsage": "",
        "ramUsed": "1.769091",
        "threadCount": "0"
    },
    "usb": {
        "USBList": [
            {
                "idProduct": 62850,
                "idVendor": 11011
            },
            {
                "idProduct": 3,
                "idVendor": 7531
            },
            {
                "idProduct": 46528,
                "idVendor": 1266
            },
            {
                "idProduct": 2603,
                "idVendor": 32903
            },
            {
                "idProduct": 151,
                "idVendor": 5002
            },
            {
                "idProduct": 2,
                "idVendor": 7531
            }
        ],
        "ZEDDetected": false,
        "errors": [
            "ZED not detected, check your USB 3.0 drivers and avoid hubs or cable extension."
        ]
    },
    "zed": {
        "ZEDCount": 0,
        "errors": [
            "ZED not detected. Connect it to an USB 3.0 port. (Blue USB ports)"
        ]
    }
}
`

The output of `lsusb`:
`
Bus 002 Device 002: ID 2b03:f582  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04f2:b5c0 Chicony Electronics Co., Ltd 
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 002: ID 138a:0097 Validity Sensors, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

adujardin, 2018-05-02:
Hi,
Yes this is a known issue with recent kernels and older ZED SDK (For more details : https://github.com/stereolabs/zed-examples/issues/53).

Psunshine, 2018-05-02:
Thanks for your quickly replay.

I noticed you said one possible solution is upgrade to CUDA 9 and use a ZED SDK >= 2.2.1.
But I just tried CUDA 9.0 and ZED SDK 2.2.1, and it reported the same problem.

Since CUDA 9.1 seems not work properly in my environment. 
Is that means I have to downgrade the kernel to 4.4?

adujardin, 2018-05-02:
Ok actually I missed it at first but it looks like you upgraded the camera firmware.

The firmware you are currently using requires the ZED SDK 2.4.0. 

So either upgrade to the latest ZED SDK or downgrade your firmware to v.1142. For the downgrade, you'll need to be able to open the ZED (using the same SDK installation to upgrade for instance)

Psunshine, 2018-05-02:
I am able to open ZED in windows 10 and I can downgrade the firmware.

Thanks very much!

Psunshine, 2018-05-02:
It finally works with firmware v.1142.
:)

lvzcl, 2018-05-10:
hello ,I have the same problem,can you tell me how to downgrade the firmware? Thanks!

lvzcl, 2018-05-11:
@adujardin can you tell me how to downgrade the firmware.Thanks

Psunshine, 2018-05-11:
First, you have to make sure that you can open ZED normally (I can access ZED in windows10).
Then, the firmware **update** option can be found in settings of **ZED Explorer**, choose the right firmware and click OK.

obraun-sl, 2018-05-11:
see here : 
https://github.com/stereolabs/zed-examples/issues/63#issuecomment-387938044

closed 2018-06-05 by Psunshine

abhijeetknayak, 2019-06-12:
Facing the same issue.
ZED SDK 2.4.0, Firmware 1523, Cuda 9
Works with windows 7 but Camera isn't detected in Linux(Ubuntu 16.04)
Also tried with SDK version 2.8, doesn't work. Any help?

shreyansh073, 2019-06-13:
> Facing the same issue.
> ZED SDK 2.4.0, Firmware 1523, Cuda 9 , Ubuntu kernel version 4.15.0-51-generic
> Works with windows 7 but Camera isn't detected in Linux(Ubuntu 16.04)
> Also tried with SDK version 2.8, doesn't work. Any help?

I think this is an issue with the kernel version. Can someone please mention the correct kernel version to be used? I am facing the same issue.

solomon-ma, 2021-06-24:
I can't put the fireware button down.
Could you tell me any other method to update the fireware to v.1142