stereolabs/zed-sdk #60: 3D Position of an object in real world

manikandtan-ck, 2018-05-07:
I need the position of an object (x,y,z coordinates in real world of chairs, refer attached image) from ZED camera in real world.
My ZED camera is stationary, placed in the corner of the room.

[image]

I already tried:
1)  getValue() from point cloud which returns the position (x,y,z) of a pixel but its not with respect to the real world and y-axis(height) of an object varies as object moves. (need the real height from camera).
(My guess is that i have to add calibration module using camera's intrinsic/extrinsic properties as well as multiple objects of known dimensions.) is this functionality already available in SDK or do I have to write one?
2) getPosition() which gives me the position of the camera with reference to WORLD_FRAME, but as my ZED is stationary how it will help?

WASCHMASCHINE, 2018-05-07:
My guess is that WORLD_FRAME only refers to the tracking module of the ZED, which means the coordinates are transformed back according to the pose estimation. For you this is irrelevant since the camera is stationary.

So, let me ask you: Where is the origin of your world coordinate system in the conference room? You need to define it and then you can try to estimate the transformation of your camera system with that origin. For instance, the camera might be up 2.2 m and is pitched down by 30° and rotated around Y by 30°. With that, you can reverse-transform a point of the ZED into your actual World Space. 

Hope this helps. Notice that this is not a ZED issue per se, but a transformation matrix chain problem. I wouldnt even call it a problem.

pollux77, 2020-04-05:
Hi,

Is a point in XYZ camera system can be transform in world space cartesian coordinate system by just do some rotation and translation transformation or I need to apply a intrinsec camera parameter to convert XYZ camera point into cartesian coordinate system ?

Oreon94-TU, 2021-04-19:
> My guess is that WORLD_FRAME only refers to the tracking module of the ZED, which means the coordinates are transformed back according to the pose estimation. For you this is irrelevant since the [...]

How can this coordinate reverse transform be done? Is there a function in pyzed. sl that allows the reverse transform using the IMU rotation values?

obraun-sl, 2021-04-20:
Hi,

You need to use RuntimeParameters::measure3D_reference_frame in world to output metrics in real world (otherwise they are in camera frame).
See here : https://www.stereolabs.com/docs/api/structsl_1_1RuntimeParameters.html#a95f5726cd4db955a263d913d310f38e4

closed 2022-04-26 by github-actions[bot]

duy1851999, 2024-10-12:
Hi there, how about estimate the height from the red points to ground, I know the height, roll, pitch, yaw  of ZED 2i camera