stereolabs/zed-sdk #61: ZED Depth Viewer and ZED Calibration seg-faulting

ajdroid, 2018-05-07:
Both the above tools are seg faulting while I run them. I have run ZED Diagnostics and all tests are OK.
Diagnostic log is below:
> {
>     "ZED SDK Components": {
>         "DiagnosticZEDSDKVersion": "2.2.1"
>     },
>     "cuda": {
>         "CUDA Toolkit version": "V9.0.176",
>         "SMI": "(SMI) :     Performance State               : P8",
>         "deviceCount": 3,
>         "deviceDriverVersion": 9010,
>         "devices": [
>             {
>                 "arch": "Pascal",
>                 "computeCapability": "6.1",
>                 "cores": 3584,
>                 "name": "GeForce GTX 1080 Ti",
>                 "totalMemoryBytes": "4294967295",
>                 "totalMemoryMB": 4096
>             },
>             {
>                 "arch": "Pascal",
>                 "computeCapability": "6.1",
>                 "cores": 3584,
>                 "name": "GeForce GTX 1080 Ti",
>                 "totalMemoryBytes": "4294967295",
>                 "totalMemoryMB": 4096
>             },
>             {
>                 "arch": "Pascal",
>                 "computeCapability": "6.1",
>                 "cores": 3584,
>                 "name": "GeForce GTX 1080 Ti",
>                 "totalMemoryBytes": "4294967295",
>                 "totalMemoryMB": 4096
>             }
>         ],
>         "initResult": 0,
>         "loaded": true
>     },
>     "hardware": {
>         "OS": "Ubuntu 16.04.4 LTS ",
>         "PCName": "bigfoot",
>         "avxSupported": true,
>         "coreCount": "12",
>         "cpu": " Intel(R) Core(TM) i7-6850K CPU @ 3.60GHz",
>         "datetime": "2018-05-07 05:49:50 PM",
>         "hyperThreading": "1",
>         "ramCapacity": "67.392557",
>         "ramUsage": "",
>         "ramUsed": "66.976707",
>         "threadCount": "24"
>     },
>     "usb": {
>         "USBList": [
>             {
>                 "idProduct": 3,
>                 "idVendor": 7531
>             },
>             {
>                 "idProduct": 2,
>                 "idVendor": 7531
>             },
>             {
>                 "ZEDUSBMode": 3,
>                 "bDescriptorType": 1,
>                 "bDeviceProtocol": 1,
>                 "bLength": 18,
>                 "bMaxPacketSize0": 9,
>                 "bNumConfigurations": 1,
>                 "bcdDevice": "1.0",
>                 "bcdUSB": "3.0",
>                 "bcdUSBClass": 239,
>                 "bcdUSBSubClass": 2,
>                 "iManufacturer": 1,
>                 "iProduct": 2,
>                 "iSerial": 0,
>                 "idProduct": 62848,
>                 "idVendor": 11011
>             },
>             {
>                 "idProduct": 528,
>                 "idVendor": 1115
>             },
>             {
>                 "idProduct": 3,
>                 "idVendor": 7531
>             },
>             {
>                 "idProduct": 8736,
>                 "idVendor": 1367
>             },
>             {
>                 "idProduct": 521,
>                 "idVendor": 1115
>             },
>             {
>                 "idProduct": 2,
>                 "idVendor": 7531
>             },
>             {
>                 "idProduct": 32770,
>                 "idVendor": 32903
>             },
>             {
>                 "idProduct": 2,
>                 "idVendor": 7531
>             },
>             {
>                 "idProduct": 32778,
>                 "idVendor": 32903
>             },
>             {
>                 "idProduct": 2,
>                 "idVendor": 7531
>             }
>         ],
>         "ZEDDetected": true,
>         "ZEDUSBMode": 3
>     },
>     "zed": {
>         "ZEDCount": 1,
>         "deviceInitResult": true,
>         "firmware": 1142,
>         "resolutions": {
>             "HD1080@15": {
>                 "fps (input)": 15,
>                 "fps (output)": 15,
>                 "height (input)": 1080,
>                 "height (output)": 1080,
>                 "initialized": true,
>                 "width (input)": 1920,
>                 "width (output)": 1920
>             },
>             "HD1080@30": {
>                 "fps (input)": 30,
>                 "fps (output)": 30,
>                 "height (input)": 1080,
>                 "height (output)": 1080,
>                 "initialized": true,
>                 "width (input)": 1920,
>                 "width (output)": 1920
>             },
>             "HD2K@15": {
>                 "fps (input)": 15,
>                 "fps (output)": 15,
>                 "height (input)": 1242,
>                 "height (output)": 1242,
>                 "initialized": true,
>                 "width (input)": 2208,
>                 "width (output)": 2208
>             },
>             "HD720@15": {
>                 "fps (input)": 15,
>                 "fps (output)": 15,
>                 "height (input)": 720,
>                 "height (output)": 720,
>                 "initialized": true,
>                 "width (input)": 1280,
>                 "width (output)": 1280
>             },
>             "HD720@30": {
>                 "fps (input)": 30,
>                 "fps (output)": 30,
>                 "height (input)": 720,
>                 "height (output)": 720,
>                 "initialized": true,
>                 "width (input)": 1280,
>                 "width (output)": 1280
>             },
>             "HD720@60": {
>                 "fps (input)": 60,
>                 "fps (output)": 60,
>                 "height (input)": 720,
>                 "height (output)": 720,
>                 "initialized": true,
>                 "width (input)": 1280,
>                 "width (output)": 1280
>             },
>             "VGA@100": {
>                 "fps (input)": 100,
>                 "fps (output)": 100,
>                 "height (input)": 376,
>                 "height (output)": 376,
>                 "initialized": true,
>                 "width (input)": 672,
>                 "width (output)": 672
>             },
>             "VGA@15": {
>                 "fps (input)": 15,
>                 "fps (output)": 15,
>                 "height (input)": 376,
>                 "height (output)": 376,
>                 "initialized": true,
>                 "width (input)": 672,
>                 "width (output)": 672
>             },
>             "VGA@30": {
>                 "fps (input)": 30,
>                 "fps (output)": 30,
>                 "height (input)": 376,
>                 "height (output)": 376,
>                 "initialized": true,
>                 "width (input)": 672,
>                 "width (output)": 672
>             },
>             "VGA@60": {
>                 "fps (input)": 60,
>                 "fps (output)": 60,
>                 "height (input)": 376,
>                 "height (output)": 376,
>                 "initialized": true,
>                 "width (input)": 672,
>                 "width (output)": 672
>             }
>         },
>         "serial": 15097,
>         "videoNumber": 0
>     }
> }

adujardin, 2018-05-09:
Hi,
Could you contact the support at support@stereolabs.com for this issue?

ajdroid, 2018-05-09:
Done, thanks

closed 2018-05-09 by ajdroid