stereolabs/zed-sdk #62: Intrinsics of depth map.

JackHunt, 2018-05-08:
Hi,
Is there a way to extract the projection matrix C from the depth map produced by the ZED SDK? 

This would be useful to use the Zed as a drop in replacement for sensors such as the PrimeSense, where the projection matrix is required to raycast in to a volume such as a TSDF.

Best regards,
Jack

WASCHMASCHINE, 2018-05-08:
@JackHunt: Yes, if you know the used calibration file (*.conf). Then you can backproject with cx,cy,fx,fy and the depth value. 

Also notice that a depth map has no intrinsics, but only the camera model has extrinsical and intrinsical parameters.

JackHunt, 2018-05-09:
Ah yes, thank you. I wasn't sure if the depth map properties would be defined by the sensor intrinsics and thus require an additional projection matrix.

closed 2018-05-09 by JackHunt