stereolabs/zed-sdk #63: TX2 can not recognize ZED

ArtlyStyles, 2018-05-09:
Hi, I have ZED connected to my Jetson TX2 with Jetpack 3.1. But when I do "lsusb", I got the following:

Bus 002 Device 002: ID 2b03:f582  

Also, ZED Explorer stayed in "Stand by" mode and did not show image. The "id" and "SN" fields on the top right corner of the GUI were empty. How can I fix this? I am using https://www.stereolabs.com/developers/downloads/archives/ZED_SDK_Linux_JTX2_v2.2.1.run

Thanks,

obraun-sl, 2018-05-09:
Hi,
Your ZED has the new firmware (v1523 since the pid changes to f582). Please use the ZED SDK 2.4.0 with this firmware.

If you want to use the old firmware, then you need to downgrade the firmware using ZED Explorer in v2.4.0 (and use the v1142)

lvzcl, 2018-05-10:
@obraun-sl 
Hi,can you tell me how to downgrade the firmware,Thanks!

obraun-sl, 2018-05-11:
**How to downgrade firmware from 1523 to 1142 :** 

* Install a ZED SDK v2.4 version (or up) on a Linux or Windows PCs (not Jetson)
* Open ZED Explorer with the camera with fw 1523 connected. The camera must open properly to update the firmware
* Click on settings button on the top right (wheel icon). Go to the firmware tab. Click on update and navigate to the specific folder for the camera : 
           -- ZED : <ZED SDK Root folder>/firmware/ZED/
           -- ZED-M : <ZED SDK Root folder>/firmware/ZED-M/
* Select the v1142 bin file (firmware 1142) and click OK.
* Wait for the process to end (it can write the new firmware multiples times, don't disconnect before the progress bar disappears)
* Unplug and plug again the camera and check that the camera works with new/old firmware
* Re-install a previous ZED SDK version

Idision88, 2018-06-23:
@obraun-sl I followed the step you mentioned above,now it stays at 99% for ten minutes,is that a right conditon?What should I do next.

lanrf11, 2019-07-29:
> **How to downgrade firmware from 1523 to 1142 :**
> 
>     * Install a ZED SDK v2.4 version (or up) on a Linux or Windows PCs (not Jetson)
> 
>     * Open ZED Explorer with the camera with fw 1523 connected. The camera must open properly to update the firmware
> 
>     * Click on settings button on the top right (wheel icon). Go to the firmware tab. Click on update and navigate to the specific folder for the camera :
>       -- ZED : /firmware/ZED/
>       -- ZED-M : /firmware/ZED-M/
> 
>     * Select the v1142 bin file (firmware 1142) and click OK.
> 
>     * Wait for the process to end (it can write the new firmware multiples times, don't disconnect before the progress bar disappears)
> 
>     * Unplug and plug again the camera and check that the camera works with new/old firmware
> 
>     * Re-install a previous ZED SDK version

Helpful!

closed 2021-12-01 by bot-stereolabs