stereolabs/zed-sdk #64: ZED SVO Recording Error

ajdroid, 2018-05-09:
Hi, I'm trying to run the ZED_SVO_Recording example. I get this:

> ZED (Init) >> Video mode: HD720@60
> ZED (Init) >> No calibration file found. Downloading it... 
> calibration file downloaded.
> Recording initialization error. Error code: SVO Recording Error
>  Note : This error mostly comes from a wrong path or missing writing permissions.

It works when I use `sudo`, so it's probably a permissions issue. I have write permissions to the directory I'm trying to dump the SVO to. What other permissions could be missing?

adujardin, 2018-05-11:
Hi,
You need to specify the filename with the extension ".svo"
Something like this : 

```sh
./ZED_SVO_Recording ./file.svo 
```

ajdroid, 2018-05-11:
Ah, that worked. I was not giving it an extension. I've fixed it to handle this error better locally, but I'm using version 2.2. Dunno if a PR will help you. Thanks!

closed 2018-05-11 by ajdroid

CircleOnCircles, 2021-09-28:
> Hi,
> You need to specify the filename with the extension ".svo"
> Something like this :
> 
> ```shell
> ./ZED_SVO_Recording ./file.svo 
> ```

similar to this solution, in my case i find the extension need to be `.svo` only