stereolabs/zed-sdk #66: can not find function isValidMeasure()

Leonslam, 2018-05-10:
I have two questions when using zed with sdk.

First, I can grab image_1 with the param "VIEW_LEFT" and image_2 with the param "VIEW_LEFT_UNRECTIFIED", which image's pixels is aligned with the depth image(data)?

Second, how can I judge  if the depth image's value is valid? According to the overview (https://docs.stereolabs.com/overview/depth-sensing/advanced-settings/) I can use isValidMeasure() but I can not find it in the header files.

obraun-sl, 2018-05-11:
Hi,

* Image with VIEW_LEFT is the one aligned with the depth data/image. the unrectified is the raw image coming from the camera and therefore not rectified and undistorted.

* isValidMeasure() is located in sl_zed/defines.hpp. It basically calls std::isfinite() to check if the pixel is valid.

Best,

Leonslam, 2018-05-14:
Many thanks!

closed 2018-05-14 by obraun-sl