stereolabs/zed-sdk #68: grab() cannot grab the newest image

Leonslam, 2018-05-16:
I wrote a demo like this:

Camera zed;
some configuration parameters;
zed.open();
while(1)
{
        zed.grab();
        get gray image and depth image;
        some processing
}

I found that the image obtained by grab function is not the newest when processing is time-consuming(like delaying 0.5s just for test). So how can I get the newest image when my processing is time-consuming? Multithreading may work but I still want your professional help. Thank you.

mpperez, 2018-07-13:
Hi,
I have an application(linux) runnig a second thread only for grab() and is fine if the cpu is not heavy loaded. 
But if the cpu is processing a lot of information the fps from the camera slow down also.
Althoug the grabbing thread could be a temporary solution, is a waste of cpu usage for small embedded devices.
Software trigger or grablastframe()  would be very interesting for low cpu devices. Also the cappability of adding a ROI could be very interesting.
is this issue into the next library updates roadmap?

Thanks

seba-eng, 2019-08-16:
I know this is an old question, but is there a solution meanwhile? Lets say I'm having some GUI and I want to display the current RGB and/or depth image when the user presses a button. Is it really the intended solution to call grap() all the time and waist CPU and GPU load?

obraun-sl, 2019-08-16:
you need to call grab() and this call must return SUCCESS to make sure you can retrieve the last images.

something like : 

```
#define TIMEOUT_MAX 100 //Security
void onPushButton()
{
     int timeout=0;
     while(!zed.grab() && timeout<TIMEOUT_MAX) {
            sl::sleep_ms(1);
            timeout++;
            if (timeout>=TIMEOUT_MAX)
                  return; //something wrong        
      }

       zed.retrieveImage(img);
       //Do your stuff.

}
```

seba-eng, 2019-08-19:
I'm using the python api and it always takes me several successful grabs until I get a new image. 
Let's say I move my hand in front of the camera: I would need to grab a new image multiple times before I see the hand in the picture. 
Example code:
```python
zed = sl.Camera()

init_params = sl.InitParameters()
init_params.camera_resolution = sl.RESOLUTION.RESOLUTION_VGA  
init_params.camera_fps = 30 

err = zed.open(init_params)
if err != sl.ERROR_CODE.SUCCESS:
    exit(1)

runtime_parameters = sl.RuntimeParameters()

def get_image():
    image = sl.Mat()
    if zed.grab(runtime_parameters) == sl.ERROR_CODE.SUCCESS:
        zed.retrieve_image(image, sl.VIEW.VIEW_LEFT)
        img = image.get_data()
        return img

```
Do you see an error there? Thanks in advance.

closed 2021-12-01 by bot-stereolabs