stereolabs/zed-sdk #71: ZED on cuda 9.2

bpinaya, 2018-05-23:
Hi there, I guess this is related to #40, but can we use ZED v2.4 with CUDA 9.2?
I need cuda 9.2 and it seems for the examples 9.1 is required. Is there anything that could be broken by that version change?
Or should I just give up and reinstall cuda 9.1

jancoow, 2018-06-15:
Hi,
I encountered the same problem. However, as far as I know the ZED api works fine with CUDA 9.2. You just simply have to edit the cmake (which for me is located in /usr/local/zed/zed-config.cmake) and edit the cuda version.

adujardin, 2018-06-18:
Hi
You currently can't, no. It is indeed related to #40. 
You will have to wait for an official ZED SDK support for CUDA 9.2.
For now, you should reinstall CUDA 9.0 or 9.1.

closed 2021-12-01 by bot-stereolabs