stereolabs/zed-sdk #73: How to use ZED tool: SVOEditor?

Henry-HD, 2018-05-24:
Hi All, I am new in using ZED SDK. When I open ZED SVOEditor.exe, the screen like: https://drive.google.com/file/d/1SWIskHBCpf6WbcoGMj_PhcvfA6Nwgf5u

If I type any key then the screen close.  I do not how to use ZED SVoEditor.exe.
System: Windows10, VisualStudio2015, CUDA8.0, ZED SDK2.2

It is the same for example "export" at this link (https://github.com/stereolabs/zed-examples/tree/master/svo%20recording/export). I use example code and follow guide "build" (no error), release, and "start". I get this command window screen: 
https://drive.google.com/file/d/1mf2-m0EEVxVpryHXl8voRzr3Zp-pVEc8

I type as instruction:
ZED_SVO_Export "C:\Users\Alibaba\Documents\ZED\VGA_SN11811_19-00-29.svo" "C:\Users\Alibaba\Documents\ZED\video.avi" 0
 
The command window close immediately and I get nothing. the SVO file is recorded in LOSSY mode.
Could you please show me where is wrong and how to fix it
Any help is appreciated! My email: blueseakid@gmail.com
Thank in advance

obraun-sl, 2018-05-25:
Hi,
Both SVOEditor and sample must be used in command line. Therefore you cannot just double click on the exe file.
When you are on the folder where the exe file is located (tools or build/Release...), type cmd in the top bar of explorer window to open a command window at this specific location. Then type the instructions as indicated 
( `ZED_SVO_Export.exe "C:\Users\Alibaba\Documents\ZED\VGA_SN11811_19-00-29.svo" "C:\Users\Alibaba\Documents\ZED\video.avi" 0` for example)

Henry-HD, 2018-06-07:
Thank you for your quick response. It worked. Sorry for late reply.

closed 2018-06-07 by adujardin