stereolabs/zed-sdk #74: Invalid calibration file

rnunziata, 2018-05-28:
The calibration file is  in /usr/local/zed/settings which was downloaded when I installed the SDK. The camera is an older one.  The config file this is downloaded is SN2145.conf I can see nothing wrong with it and it matches the one that comes from the zed calibration site if I download it manually. Any thoughts on how to proceed?

```
./ZED_Tutorial_1 
ZED (Init) >> ERROR: 
## Unsupported firmware version.
## Update to the latest firmware using ZED Explorer
ZED (Init) >> Video mode: HD720@60
ZED (Init) >> Wrong value in calibration file. Downloading it... 
ZED (Init) >> Invalid calibration file. Recalibrate your camera using ZED Calibration tool or download your calibration file on https://calib.stereolabs.com
```

adujardin, 2018-05-29:
`## Unsupported firmware version` means that you have to upgrade your ZED firmware to be able to use the ZED SDK and the associated calibration file.

You should upgrade it using ZED Explorer : Settings (Gear on the top right) -> Firmware -> Update.

rnunziata, 2018-06-03:
upgraded firmware but also had to manually calibrate camera using calibration tool  . The downloaded calibration file format is no longer supported my camera is over three years old. After that camera and tutorials , samples work fine.

closed 2018-06-03 by rnunziata