stereolabs/zed-sdk #76: Unable to compile sample examples

NihalKarne, 2018-06-04:
As given in the document i followed the steps to compile the examples.[https://docs.stereolabs.com/overview/getting-started/application-development/](url)
But when is fire make command i get these errors : 
```Scanning dependencies of target ZED_Tutorial_1
[ 50%] Building CXX object CMakeFiles/ZED_Tutorial_1.dir/main.o
[100%] Linking CXX executable ZED_Tutorial_1
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaFree@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaBindTexture2D@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_32s@libnpps.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMallocPitch@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_8u@libnpps.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemset2DAsync@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaGetErrorString@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaStreamSynchronize@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaDriverGetVersion@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `nppiFilterBoxBorder_8u_C4R@libnppif.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemset3D@libcudart.so.9.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `cudaMallocManaged@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiFree@libnppisu.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemcpy@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemcpyToSymbolAsync@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaGetDeviceCount@libcudart.so.9.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `cudaHostAlloc@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaEventRecord@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `nppiSet_32f_C1R@libnppidei.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaStreamCreate@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiFilterBoxBorder_8u_C1R@libnppif.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaConfigureCall@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiMalloc_32f_C1@libnppisu.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemGetInfo@libcudart.so.9.0'
/usr/local/zed/lib/libsl_tracking.so: undefined reference to `nppiMalloc_8u_C1@libnppisu.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiMirror_8u_C4IR@libnppig.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiHistogramRangeGetBufferSize_32f_C1R@libnppist.so.9.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `cudaMemcpy2D@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaGetDeviceProperties@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaCreateChannelDesc@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaStreamDestroy@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaDeviceReset@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaStreamAddCallback@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsFree@libnpps.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaEventCreateWithFlags@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaGetLastError@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaSetDevice@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiMulC_32f_C1IR@libnppial.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiHistogramRange_32f_C1R@libnppist.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaMemcpyToSymbol@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `nppiAdd_32f_C1R@libnppial.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMalloc3D@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaMalloc@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `__cudaRegisterTexture@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_32f@libnpps.so.9.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `cudaFreeHost@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `__cudaRegisterFatBinary@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemcpy2DAsync@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaUnbindTexture@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `__cudaInitModule@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaMemcpyAsync@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `__cudaUnregisterFatBinary@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaSetupArgument@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaMemset2D@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `__cudaRegisterFunction@libcudart.so.9.0'
/usr/local/zed/lib/libsl_depthcore.so: undefined reference to `cudaRuntimeGetVersion@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `__cudaRegisterVar@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaDeviceSynchronize@libcudart.so.9.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cudaLaunch@libcudart.so.9.0'
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_Tutorial_1.dir/build.make:115: recipe for target 'ZED_Tutorial_1' failed
make[2]: *** [ZED_Tutorial_1] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_Tutorial_1.dir/all' failed
make[1]: *** [CMakeFiles/ZED_Tutorial_1.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
Ubuntu 16.04 LTS 
CUDA Version 9.0.176 
Kernel version 4.13.0-43-generic
SDK version 2.2.1

How to solve these errors ?

canberkgurel, 2019-06-22:
@NihalKarne @ravikt Did you manage to solve this issue? My SDK is already updated to the latest version but I am still getting the exact same error.

ravikt, 2019-06-23:
Hello @canberkgurel , are you trying to install on Jetson or Linux desktop ?
Please check the corresponding version of Ubuntu and CUDA. For Ubuntu 16 both CUDA 9 and 10 are compatible with SDk 2.8.2. You have to see which one works for you.

canberkgurel, 2019-07-01:
@ravikt I was installing it to a Linux Laptop running Ubuntu 16.04. I installed CUDA 10 and SDK 2.8.2. But I don't have the permission to disable the secure boot - some drivers may not be installed properly.?

ravikt, 2019-07-03:
@canberkgurel If it is not Jetson then it should have installed without any problem. Can you try with CUDA 9 ? But you have to be careful when you remove CUDA 10. I am not sure if it will work but you should give it a try.

closed 2022-04-26 by github-actions[bot]