stereolabs/zed-sdk #79: Problem with XYZ data during an acquisition

Lemany, 2018-06-13:
Hello everybody,

I am a biomedical student and I use the ZED camera on a project that deals with the analysis of walking.
To explain the context, I use a treadmill and a patient will go on it.
We have a few LEDs symmetrically on the legs to analyze the walk.

By some image processing, I take back the barycentre of each LED and thus the 2D coordinates of each LED. Via the functions of SDK

, I also get the coordinates X, Y, Z.

Ok very good, when I want to analyze this recorded data. I notice a big problem, there is an asymmetry between the two legs on the Y and Z axes.

Here are two screenshots, in this one obtained via another system of acquisition, no problem of asymmetry. : [image]

Here is the data retrieved with the ZED: [image](the Y and Z axes are reversed)

I hope someone could find a track for this problem.

Regards,

WASCHMASCHINE, 2018-06-14:
Hi @Lemany 

So plotting only the tracked LED Y (or whatever is vertical) in image space would work to fix the asymmetry? If you could fix the asymmetry in image space it will most likely look correct translated to Euclidean space. You could then do it manually via backprojection (4 lines of code) and not use that particular SDK function.

What are your runtime parameters and general settings for the ZED? I suppose if you have tracking enabled, things could diverge in a static scene. But the latest tracking is quite good.

Lemany, 2018-06-14:
Hi @WASCHMASCHINE 

Thank you for your quick response.

I can show you the 3D rendering  : 

[image]

As you can see on the top of the body (Asis and Thigh) there is no problem. 
But for the shank and toe, we can see there is a big problem with the depth distance, it's very strange.

I will try to translate to Euclidean space.

For the camera paramaters, I will add it when I can .

WASCHMASCHINE, 2018-06-14:
I do not know what you mean with "aligned data".

I also cannot see what you mean in your plot. What exactly is the problem? I would have assumed that maybe the toes have a symmetric trace, correct?

You did not answer my question whether plotting the pixel in 2D would yield symmetry. Thus we cannot exclude a non ZED error.

WASCHMASCHINE, 2018-06-14:
Also, how is the camera situated to the person? Like how far is the closest point away? Too close objects may not be seen by both cameras or too high disparities are excluded from the computation for performance.

Lemany, 2018-06-14:
Zed SDK allows only to retrieve data(RGB,DEPTH etc)  from Left camera or Right but not both.

My problem is as the left foot does not pass in front of the right foot and also the right foot never passes in front of the left foot. Normally, when you walk that's what happen.
About the trace, it's correct no problem, juste the position between the toes is not correct.
 You can compare the profil and the previous graph. It's like an offset, I don't know ..

Yes in 2D there is this symmetry, I can show you pictures by private message if you want.

The distance between the camera and the person is like 2meters. I think it's better if I show you my recording pictures, to be more comprehensive.

WASCHMASCHINE, 2018-06-14:
The SDK allows you to retrieve images for almost anything (Enumeration VIEW): 
https://www.stereolabs.com/developers/documentation/API/v2.4.0/group__Video__group.html
And depth stuff (Enumeration MEASURE):
https://www.stereolabs.com/developers/documentation/API/v2.4.0/group__Depth__group.html

My advice would be just to use retrieveMeasure with MEASURE_DEPTH and find the associated depth for the tracked pixel. Then, backproject the UVZ back to XYZ manually.

Backprojection like this, just reverse it:
https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html
x = (u-cx)/fx*z
y = (v-cy)/fy*z
z = z    :smile:

WASCHMASCHINE, 2018-06-14:
BTW retrieveImage with VIEW_DEPTH will only be a graphical representation of the depth map for visualization purposes.

Another idea: Maybe the camera is situated on the height of the hip and you are somehow using distances instead of depths? IDK

Lemany, 2018-06-14:
Ok I will try that.

But should I retrieve : MEASURE_DEPTH  or  MEASURE_DEPTH_RIGHT  ? 
I'm working on the RGB picture from Right camera.

Currently i'm using MEASURE_XYZRGBA, 
I think maybe the error is as I'm not using MEASURE_XYZRGBA_RIGHT  !

WASCHMASCHINE, 2018-06-14:
This could be it. Use the *_RIGHT version when you use VIEW_RIGHT for the LED tracking image.

Lemany, 2018-06-14:
Yes I think my error is from that. Anyway I will modify it and check if it's correct. 
Thx very much @WASCHMASCHINE !

WASCHMASCHINE, 2018-06-14:
Let me know if it fixes it!

Lemany, 2018-06-14:
Ok no problem, tomorrow I will get new results

Lemany, 2018-06-15:
I try with this code but all of my data are equal to 0 ...

` zed2.retrieveImage(image_zed2, VIEW_RIGHT, MEM_CPU, new_width2, new_height2);
            zed2.retrieveImage(depth_image_zed2, VIEW_DEPTH_RIGHT, MEM_CPU, new_width2, new_height2);
            // Récupérer le nuage de points RGBA en demi-résolution
            zed2.retrieveMeasure(point_cloud2, MEASURE_XYZRGBA_RIGHT, MEM_CPU, new_width2, new_height2);`

obraun-sl, 2018-06-15:
Hi,

If you are using right measures (depth or xyz), don't forget to enable the "right side measurements" in initParameters, since they are not activated by defaults (because this requires more memory and time).

`init_parameters.enable_right_side_measure = true;
`

Lemany, 2018-06-15:
Hi @obraun-sl  , thank you for this advice because I ignored that need this activation.
I will try this right now.

Can you explain me, what VIEW_DEPTH and  MEASURE_XYZRGBA return ?
For me it's not clear, because we can choose only left or right side for the RGB and then we have some VIEW_DEPTH where we can choose 'Left','Right' or none.

Regards,

obraun-sl, 2018-06-15:
Every calculated data (depth, point cloud (xyz), tracking) is always by default referring to the Left sensor/Left image. That is why we call that VIEW_DEPTH, MEASURE_XYZRGA, but it is mapped on the left image. VIEW_DEPTH will show the depth view for left image, MEASURE_XYZRGBA the point cloud data for left image as well.
( That's right, we could have called it VIEW_DEPTH_LEFT and MEASURE_XYZRGBA_LEFT, but on early version of the ZED SDK, the right side was not available and left side was the default and only side available. Therefore, to avoid breaking the API, we decided to keep the same name).

Note : For images, since left and right images are the basic components for calculating all the data, they are always available. But since right side measurements require some time to calculate, then it is not enabled by default (only some applications such as AR requires the depth for both sides).

Second Note : VIEW_DEPTH will provide uchar (0-255) data, therefore it should only be used for viewing purposes but not for depth data extraction. MEASURE_XYZRGBA(_RIGHT) will provide the point cloud data in 4 channels of float.

Lemany, 2018-06-15:
Thank your for response, it's very clear for me now !

Before I was retrieving like that:
`
zed.retrieveImage(image_zed, VIEW_LEFT, MEM_CPU, new_width, new_height);
zed.retrieveImage(depth_image_zed, VIEW_DEPTH, MEM_CPU, new_width, new_height);
zed.retrieveMeasure(point_cloud, MEASURE_XYZRGBA, MEM_CPU, new_width, new_height);
}`
But as you can see my resultats are strange, as an offset between the retrieve image. So now, i'm taking all of the datas from the right, normally it will correct my problems, I hope.

WASCHMASCHINE, 2018-06-15:
@Lemany Backproject the pixel with the mentioned formulas and then plot the axis of the 3D point.

Lemany, 2018-06-15:
That's the same results when I retrieve MEASURE_DEPTH and apply the back projection :/

WASCHMASCHINE, 2018-06-15:
So you are telling me that in image space the pixels behave normally but when you just backproject them along the ray they will somehow lose it? Have you tried setting the depth to 2 m for both? There is no way that it would behave differently than in image space. Maybe you mixed up coordinate spaces.

Lemany, 2018-06-20:
Hi @WASCHMASCHINE ,

I did a lot of test. So finally, We have got like a distorsion because the angle of ZED camera is not perpendicular in front of the body. 
So we have a distorsion, and that explains the offset between the top of the legs and the toes, there is a distorsion. There is a way to correct it ?

Regards,

WASCHMASCHINE, 2018-06-20:
@Lemany 
First of all, the camera is pitched down or what? Then you can just compute the 3D points and move to a coordinate system with adjusted pitch.

Here are some checks you can do anyway, but might not help:

1. Do you use the rectified versions of the images? Raw images are distorted.
2. Have you tried calibrating the camera (using the calibration tool) and redid the test with new calibration file?
3. Are you positive that you are not computing distances somehow instead of depth? This would mean that the center of the images looks normal, but the points away from it will be more distant. So, what happens if you set the depth to 2 meters?

closed 2022-04-26 by github-actions[bot]