stereolabs/zed-sdk #81: ZED tools

jancoow, 2018-06-15:
Hi there,

I'm mainly using Arch linux for the ZED development and everything works fine. However, I want to calibrate my camera and I need the calibration tool for that. But I can't find the source code of these tools, so I can't compile them for my platform. Where is this source code located?

adujardin, 2018-06-18:
Hi,
If you want to calibrate your camera you can use the tool  `ZED Calibration` in `/usr/local/zed/tools/`.
We do not provide the tools source code, you will have to use the compiled version (by installing the correct dependencies in your case). 
Please note that we only provide support for Ubuntu 16.04 for the current ZED SDK version, any tweaks to make it work on other distribution is up to you.

bitshiftnetau, 2019-08-12:
I too would like a response to this question. There does not seem to be any reasonable explanation available. One would expect that a development tool such as the ZED camera should have an open source development tool available. If not the 3D stitcher, then at least the calibration and configuration tools. 

Surely, your only defensible position can't possibly be 'it's closed source'. It does not seem reasonable to place the burden of development solely on the ZED team when there is no actual benefit to the user. 

I certainly did not spend 800AUD on a toy, to have these questions go unanswered at the very least.

closed 2022-04-26 by github-actions[bot]