stereolabs/zed-sdk #86: ZED API on CUDA 9.2 ??

kooscode, 2018-07-03:
Running latest CUDA 9.2 on Ubuntu and my ZED API seems broken ???

Why are you hard linking to a specific CUDA version? 

When will you support CUDA 9.2 ??

nbiresev, 2018-07-19:
I successfully install ZED SDK on Ubuntu 17.10, with CUDA 9.2. The issue is that i can't compile samples 
because libraries like /usr/local/zed/lib/libsl_depthcore.so are missing in /usr/local/zed/lib/ folder. There are only libraries libsl_core.so  libsl_input.so  libsl_zed.so. Anyone knows why is it the case? With ZED SDK previous version install i had no issues, all libraries were there.

nesnes, 2018-07-19:
`libsl_depthcore.so` is a librairy that no more exist in the ZED SDK, only `libsl_core.so`, `libsl_input.so` and `libsl_zed.so` are expected.

The make file your are using is probably outdated, can you download and try to compile one of the tutorial? https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%201%20-%20hello%20ZED

nbiresev, 2018-07-19:
Why would i have outdated makefile, considering that i was trying to compile ZED 2.5 samples that come with installation of ZED SDK? I have tried your example also, it doesn't work, the same error: 

[ 50%] Building CXX object CMakeFiles/ZED_Camera_Control.dir/src/main.o
make[2]: *** No rule to make target '/usr/local/zed/lib/libsl_depthcore.so', needed by 'ZED_Camera_Control'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_Camera_Control.dir/all' failed
make[1]: *** [CMakeFiles/ZED_Camera_Control.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

nesnes, 2018-07-19:
The libraries required by the ZED SDK are listed in `/usr/local/zed/zed-config.cmake`
On a fresh SDK 2.5 install, it should look like this:
```
...
set(ZED_LIBRARIES
        "/usr/local/zed/lib/libsl_input.so"
        "/usr/local/zed/lib/libsl_core.so"
        "/usr/local/zed/lib/libsl_zed.so"
    )
...
```

As `libsl_depthcore.so` doesn't exist anymore, you have to make sure that the issue doesn't come from older CMake files by downloading new ones, and delete the build folder of previous projects to re-create it.

nbiresev, 2018-07-19:
That is the case also for my zed-config.cmake, but still i get that error:

set(ZED_FOUND FALSE CACHE BOOL "" FORCE)

set(ZED_INCLUDE_DIRS "/usr/local/zed/include")
set(ZED_LIBRARY_DIR "/usr/local/zed/lib")

if(ZED_INCLUDE_DIRS)
    set(ZED_LIBRARIES
        "/usr/local/zed/lib/libsl_input.so"
        "/usr/local/zed/lib/libsl_core.so"
        "/usr/local/zed/lib/libsl_zed.so"
    )
    set(ZED_FOUND TRUE CACHE BOOL "" FORCE)

    exec_program(uname ARGS -p OUTPUT_VARIABLE CMAKE_SYSTEM_NAME2)
    if (CMAKE_SYSTEM_NAME2 MATCHES "aarch64" )
        unset(CUDA_USE_STATIC_CUDA_RUNTIME CACHE)
        option(CUDA_USE_STATIC_CUDA_RUNTIME OFF)
    endif()

    SET(ZED_CUDA_VERSION 9.2)

    find_package(CUDA 9.2 EXACT REQUIRED)
    if(CUDA_FOUND)
        if(NOT ${CUDA_VERSION_MAJOR} LESS 9)
            set(cuda_path "/usr/local/cuda-${CUDA_VERSION_MAJOR}.${CUDA_VERSION_MINOR}/lib64/")
            set(CUDA_NPP_LIBRARIES_ZED
                                "${cuda_path}libnppial.so"
                                "${cuda_path}libnppisu.so"
                                "${cuda_path}libnppicc.so"
                                "${cuda_path}libnppicom.so"
                                "${cuda_path}libnppidei.so"
                                "${cuda_path}libnppif.so"
                                "${cuda_path}libnppig.so"
                                "${cuda_path}libnppim.so"
                                "${cuda_path}libnppist.so"
                                "${cuda_path}libnppitc.so"
                                ${CUDA_npps_LIBRARY})
        else()
            set(CUDA_NPP_LIBRARIES_ZED ${CUDA_npp_LIBRARY})
        endif()
    endif()
endif()

nbiresev, 2018-07-19:
Removing the folder /usr/local/zed_previous/  (which was a backup installation) solves the issue.

nesnes, 2018-07-19:
Thanks for the feedback, we'll take a look in our cmake system to figure out why the previous installation was used.

# Solution
For now, removing the `/usr/local/zed_previous` folder fixes the issue.

closed 2018-07-19 by nesnes