stereolabs/zed-sdk #88: zedmini - how to get IMUData at 500hz?

axing0805091, 2018-07-05:
I notice that before using the zed.getIMUData(), we need to call zed.grab() at first. So the imu data rate is the same as the image rate, can I get imu data at 500hz?

nesnes, 2018-07-05:
getIMUData

nesnes, 2018-07-05:
Hi,
Calling getIMIData with default parameters return the IMU data synched with the image, so it will be updated at the image rate.
However, you can get the current IMU data, not synched with the image using the `sl::TIME_REFERENCE_CURRENT` parameter. This call doesn't need to be done after a grab().

```c++
sl::IMUData myData;
getIMUData(myData, sl::TIME_REFERENCE_CURRENT)
```

Here is the corresponding documentation: https://www.stereolabs.com/developers/documentation/API/v2.4.0/classsl_1_1Camera.html#a453890423c830b627fc35c63edde7615

axing0805091, 2018-07-06:
Thank you very much, it's very helpful. Can I get the image at 60hz and imu data at 500hz at the same time. 
Does the "svo recording - recording" sample save imu data at 500hz?

nesnes, 2018-07-06:
Answering in https://github.com/stereolabs/zed-examples/issues/89

closed 2018-07-06 by nesnes