stereolabs/zed-sdk #91: Saving depth map as .csv

e-delta, 2018-07-14:
I am trying to simply take the depth map and save its contents as a .csv file or other text format for the purpose of inputing into a convolution neural network. The API reference says that the type is a 1-channel MAT. How can I view these values? I have been able to return individual values using the getValue() method but I would like to save the entire matrix for multiple images.

avnishks, 2019-09-16:
@e-delta : Were you able to figure this out?

e-delta, 2019-09-19:
No, but I wrapped that project a long time ago. Don't even have the zed anymore.

obraun-sl, 2019-09-20:
For those who can be interested, here is a small code that can do that : 

`           
sl::Resolution down_res(640,360);     
zed.retrieveMeasure(depth,sl::MEASURE_DEPTH,sl::MEM_CPU,down_res.width,down_res.height);
float* rawData = depth.getPtr<float>();
std::ofstream file("depth.csv");
for (int j=0;j<depth.getHeight();j++) {
     for (int i=0;i<depth.getWidth();i++) {
         file<<rawData[i + j * depth.getWidth()]<<","; //in line progress
     }
   file<<"\n"; // end of line
 }
file.close();//done
` 
Each cell will contain the depth value (in unit specified by InitParameters) for the (i,j) pixel defined by row and column number.
Note that it can be time consuming to write the file + resolution cannot be too high due to excel limitations.

closed 2022-04-26 by github-actions[bot]