stereolabs/zed-sdk #92: Converted videos are cropped / have narrower angle of view

GRichardson90, 2018-07-18:
Hi All

I am converting videos from .svo to .avi.
The original videos were shot in 2560x720 at 60fps and the outputted videos are also 2560x720 at 60fps. However, the outputted videos appear to have a narrower angle of view. They appear cropped in, losing ~10% frame all around (See attached screenshots, most obvious when looking at plug sockets on left)
I've noticed that when recording through ZED explorer, the 1080 resolution appears to have a similar narrower angle of view, though I'm not sure this is relevent.
Please advise how I might conserve the view angle of the original videos when converted.

Thanks
G

[image]
[image]

obraun-sl, 2018-07-18:
Hi,

The SVO contains the raw images from the camera (i.e. the one you can see on ZED Explorer). 
If you are using the SVO export sample, the images used to create the avi file are the one given by the zed sdk by retrieveImage(sl::Mat, sl::VIEW_LEFT / sl::VIEW_RIGHT). Those ones are rectified, which means aligned and distortion corrected. That is why you are loosing some field of view.
If you want to keep the original field of view, then you need to use the images unrectified : 
`zed.retrieveImage(sl::Mat, sl::VIEW_LEFT_UNRECTIFIED / sl:/VIEW_RIGHT_UNRECTIFIED)`

GRichardson90, 2018-07-18:
Thanks very much for the quick response, this works!
Just to confirm, the converted, unrectified images should have the exact dimensions of the original svo's? So the ZED camera calibration parameters will be accurate and precise with respect to the converted images?

G

obraun-sl, 2018-07-18:
Yes . Unrectified and rectified have the same dimensions.
the calibration parameters given by getCameraInformation().calibration_parameters refers to the rectified images. You also have the raw calibration parameters (for unrectified images), but unrectified images are a lot more difficult to use if you plan to do 3D triangulation for example. 
It's better I think to use the rectified images.

GRichardson90, 2018-07-18:
OK, thanks for the advice, a few more last questions:
Regarding the calibration parameters displayed in the ZED explorer GUI (i.e. the same as displayed when following the link https://www.stereolabs.com/developers/calib/?SN=NNNN) - are these calibration parameters for unrectified images as shot by the camera?
If not, how can I get the calibration parameters for unrectified images?
Finally, out of interest, what is it about unrectified files that makes them more difficult to use for 3D triangulation etc.?

Thanks again
G

obraun-sl, 2018-07-19:
In ZED Explorer, it's the raw calibration parameters (unrectified images). Same that you can find on the conf file as you mentionned in the link. 
the rectification makes the images "perfect" : it means as if it was coming from perfectly identical cameras and perfectly aligned. (no rotation between them, no translation except the baseline). 
Therefore formulae for 3D triangulation (pixel to depth) becomes a lot more simple : disparity = focale * baseline / depth . If it's not rectified, this relation between disparity and depth is not this one anymore and therefore a lot more complex.

whistlefancy, 2021-09-14:
Oh, thank you everyone for answering my confusion. I used the pictures taken by the camera function to calibrate the camera, and then the converted avi video frame was used for ORBSLAM2, which could not be used normally. Therefore, I suggest that the original video will be corrected, which can be mentioned in the instructions for converting svo to avi.

closed 2022-04-26 by github-actions[bot]