stereolabs/zed-sdk #99: Zed not working after firmware update crash

dongfangstevenyang, 2018-08-21:
My ZED stereo camera is not working after a failed firmware update on Linux machine of Ubuntu.

I was trying to downgrade the firmware, but somehow the ZED SDK crashed. After that, my ZED is not working with any computer. I tried different computers of either Windows or Ubuntu with different usb ports (all usb3.0 and usb2.0). All the computers cannot recognize the ZED stereo camera (not response at all, as if the camera is not plugged in).

I guess the problem would be crashed firmware update. I searched the Internet for a whole day but can not find a solution. Could anyone help me with this issue?

nesnes, 2018-08-21:
Hi,

Can you reach support@stereolabs.com with a diagnostic file generated by the ZED Diagnostic tool?
It should help to identify the issue and initiate a replacement if needed.

nesnes, 2018-08-21:
Seems that you already reached the support, we'll run the diagnostic and replacement there.
I'm closing the issue.

closed 2018-08-21 by nesnes