stereolabs/zed-sdk #100: Bug in SpatialMapping example?

lexvandersluijs, 2018-08-29:
I'm trying to create a nice AR demo in our office with the very impressive ZED Mini + SDK, using the SpatialMapping example as a starting point. But I am experiencing some problems, hope you can help.

The idea is (running in the Editor for simplicity during development):

1. First, Spatial Memory in the ZED_Rig_Stereo gameobject is enabled,
2. The space is mapped (mesh is created) 
3. When mapping is stopped, the mesh and the .area file indeed appear in the Assets folder, great!
4. Then I add the mesh to the scene (temporarily) so that I know where to place the objects that should appear in AR. I add objects into the scene, using the mesh for guidance, so far so good.
5. Guidance mesh is deleted, since we will load it at runtime in the next step anyway

Then the idea is to go back to Play mode, load the mesh, where the ZedSpatialMapping class will also load the .area file if it exists. The tracking should try to localize itself in the point cloud, and the augmentations should appear in the right place ( I can show and hide the mesh for verification).

However, when the Load button is pressed, initially the 3D data is _saved_ first (actually there are a number of calls to the save function during the LoadMesh call, but the first one is particularly troublesome). This means that whatever we try to load is actually incorrect, if I understand correctly.

This happens because ZEDSpatialMappingManager has a an event handler (SpatialMappingHasStopped) registered on ZEDSpatialMapping.OnMeshReady, and in that handler SaveMesh() is called.

And SpatialMappingHasStopped is triggered by the call to StopSpatialMapping in "LoadMesh()" 

```
    public bool LoadMesh(string meshFilePath = "ZEDMesh.obj")
    {
        if(OnMeshStarted != null)
        {
            OnMeshStarted();
        }
        //If a spatail mapping have started, disable it
        DisableSpatialMapping();
```

I have tried to fix this issue by commenting out the call to DisableSpatialMapping(), and I think the proper data is now loaded, but the camera now often jumps between to poses (to be judged by the overlap of the 3D mesh plus my added object with the camera image), where neither of these poses are correct: one would expect that the saved 3D mesh overlaps the real world nicely when tracking has been reinitialized, but this has not been achieved yet.

I hope someone from Stereolabs can help, the mentioned intended workflow is important for us!

ckorris, 2018-08-29:
Hi, I'm Chris from Stereolabs. I'm happy to hear you're impressed with the ZED! 

For the SaveMesh() call happening when you press the Load button, my understanding is that this is a problem because it saves an empty mesh, or an existing one you no longer care about, into the path you specified for loading a mesh. That's definitely a bug, and I've logged it internally to fix it later. 

Rather than trying to avoid calls to SaveMesh (as you're correct it gets called in various situations), you can cache the value of ZEDSpatialMappingManager.saveWhenOver and set it to false at the start of LoadMesh. Then you could restore it at the end of LoadMesh so that future scans save or don't as intended. 

This should work as SpatialMappingHasStopped only calls SaveMesh() if it's set to true, and every SaveMesh() call goes through that event handler. 

The camera jumps are tougher to debug, though. I'd first see if the .area files you scan are working reliably in the regular workflow. To do this, I'd make a different Scene with ZED_Rig_Stereo or open one of the example scenes. Make sure there's some virtual objects in the scene. Then in ZEDManager, set Spatial Memory to true and specify the path to the .area file. Run the scene, move around a bit, and see if the virtual objects jump around a lot or stay positioned more reliably. Note that a bit of jumping is expected at the start as the camera localizes but it shouldn't last for long. 

This is worth testing first because localizing with spatial memory is quite difficult, so if the scan from the environment is incomplete or not detailed enough, it won't do a good job. And note that even with a good scan, there will always be imperfections. 

If the results from that test are bad, I'd try scanning differently and more thoroughly and seeing how it affects it. If you can then get the tracking to work pretty well, but can't make the same happen after LoadMesh, then there's likely a bug somewhere. 

Hope that helps!

lexvandersluijs, 2018-08-30:
Hi Chris,
Thanks for your fast response!
I implemented the tip to manipulate saveWhenOver like this, and this seems to work nicely:
```
    public bool LoadMesh(string meshPath = "ZEDMeshObj.obj")
    {
        // forcibly turn saving off, because of bug which would cause a Save during the Load
        bool oldSaveWhenOver = saveWhenOver;
        saveWhenOver = false;

        manager.gravityRotation = Quaternion.identity;

        spatialMapping.SetMeshRenderer();
        bool result = spatialMapping.LoadMesh(meshPath);

        // restore old value
        saveWhenOver = oldSaveWhenOver;
        return result;
    }
```
After this change I found these things:
- it appears that the 3D mesh that is saved is actually mirrored along the X axis. This picture here shows what the scene looks like after the Load button is pressed. 
[image]

The blue mesh is dynamically loaded and child of [ZED Mesh Holder], and the orange/shaded is the one that was loaded from disk as an asset, this was rotated 90 degrees to help me understand what was going on. When I simply apply a -1,1,1 scaling to the mesh that was loaded from disk by the Unity editor, it overlaps the one that is loaded by the ZED SDK perfectly.

Of course, this means that I was placing my 3D objects for AR in the wrong place in the scene. So after correcting my 'guidance' mesh, I tried again. 
No success yet though, the 3D objects still appeared in incorrect locations.

- I added a label in ZEDSpatialMappingManager.OnGUI to show the tracking state
```
    public void OnGUI()
    {
        GUI.Label(new Rect(20, 20, 500, 100), "Tracking state: " + manager.trackingState);

    }
```

While the tracking starts it quickly goes to TRACKING_OK. But after pressing the Load button, which loads the pre-mapped point cloud from the .area file, the tracking goes to TRACKING_SEARCH, no matter where I look with the camera. A picture of a part of the scene that was mapped is below, to indicate that the scene should have a good amount of recognizable feature points.
[image] 

So it seems that the camera is not actually relocalized, which explains why the augmentations are not appearing in the correct spot (they are _approximately_ correct, but only because the starting pose of the camera is similar from run to run).

Hope this helps with troubleshooting!

lexvandersluijs, 2018-08-30:
Update: after a few more runs I find that in some cases the tracking state does go to TRACKING_OK :)

Having said that, the actual relocalization quality is quite variable. Sometimes it is _spot on_ (verified by enabling mapping again, and seeing how well the new mesh overlaps with the pre-made one), but sometimes it is really off as well (+10 degrees of rotation, a meter of translation). Is there a way to set a 'confidence' threshold somewhere (e.g. based on reprojection error, number of features matched, or other (combined) metrics)? Perhaps that could help against 'false positives'. 

I've also seen the 'flickering' result again, where the pose jumps back and forth (hypothesis: could it be that there are two point clouds in memory, one generated on the spot and one that is in memory, and the system can't decide between the two?).

ckorris, 2018-08-31:
Thanks for verifying that the saveWhenOver fix worked! That's the fix I planned for the plugin, so now I can do it with more confidence. :) 

For the relocalization, one of our R&D guys took a look at the issue and said it's a known issue, and it's indeed related to confidence as you mentioned. We actually have a confidence check internally, but we set it pretty loose. We'll have to first make this a parameter in our SDK (and soon after, Unity) to make it adjustable. I'll add it to our to-do list. 

Though in the meantime, you should try turning on pose smoothing in Unity to see if that helps make the jumps less dramatic. If you look at the EnableTracking() function in ZEDCamera, you'll see it has an enablePoseSmoothing argument that defaults to false. None of the three places in the plugin it's called (including in LoadMesh) change the value. You could set it to true to see if it helps. 

Though I'm not positive why it defaults to false - looking into it. But should be worth trying.

closed 2022-04-26 by github-actions[bot]