stereolabs/zed-sdk #106: Does ZED SDK 2.6 support  RTX 2080 Ti (Turing)?

hchoi405, 2018-10-19:
I installed new RTX 2080 Ti, and executed zed depth viewer which was successful with GTX 1080 Ti.
But it says "GPU memory limit reached".
I attach the part of the diagnostic tool result.

```json
    "Camera Test": {
        "ZEDCount": 1,
        "deviceInitResult": 0,
        "firmware": 1523,
        "serial": 18563,
        "valid": [
            "<b>Camera: </b>ZED",
            "<b>Firmware: </b>1523"
        ],
        "videoNumber": 0
    },
    "Graphics Card": {
        "deviceCount": 1,
        "deviceDriverVersion": 10000,
        "devices": [
            {
                "arch": "unknown",
                "computeCapability": "7.5",
                "cores": 0,
                "name": "GeForce RTX 2080 Ti",
                "totalMemoryBytes": "4294967295",
                "totalMemoryMB": 4096
            }
        ],
        "initResult": 0
    },
    "Processor": {
        "OS": "Windows 10 FallCreators",
        "avxSupported": true,
        "coreCount": "12",
        "cpu": "Intel(R) Xeon(R) CPU E5-2687W v4 @ 3.00GHz",
        "datetime": "2018-10-19 13:11:11",
        "hyperThreading": "1",
        "motherboard": "X99 Professional Gaming i7, ASRock",
        "ramCapacity": "63.9207",
        "ramUsage": "0",
        "ramUsed": "4.99728",
        "threadCount": "24",
        "valid": [
            "<b>Processor: </b> Intel(R) Xeon(R) CPU E5-2687W v4 @ 3.00GHz",
            "<b>Motherboard: </b> X99 Professional Gaming i7, ASRock"
        ]
    },
```

Is there any problem with my setting?
Or just the ZED SDK does not support the new GPU?

I think this is the latter case because the size of memory above is wrong,,

hchoi405, 2018-10-19:
Some missing info
```json
"ZED SDK": {
        "CUDA Toolkit version": "V9.2.148",
        "CUDA_reachable": true,
        "DiagnosticZEDSDKVersion": "2.6.0",
        "InstalledZEDSDKVersion": "2.6.0",
        "ZEDSDKFound": true,
        "driverVersion": "416.34",
        "drivervDllVersion": "25.21.14.1634",
        "initResult": 0,
        "loaded": true,
        "runtimeDllVersion": "",
        "valid": [
            "<b>ZED SDK version:</b> 2.6.0"
        ]
    }
```

adujardin, 2018-10-19:
No the ZED SDK 2.6 doesn't support Turing cards, the next release ZED SDK 2.7 with CUDA 10 support will. It's currently in QA and should be out around next week.

adujardin, 2018-10-26:
The 2.7 is now released (https://www.stereolabs.com/developers/release/2.7/)

closed 2018-10-26 by adujardin