stereolabs/zed-sdk #107: ZED SDK on Jetpack 3.3 for TX2

mprat, 2018-10-19:
I am trying to benchmark the ZED camera for our depth sensing application on a Jetson TX2 board. However, I am not able to get my Jetson TX2 dev board to initialize the camera. I am running Jetpack 3.3, and the camera is connected directly into the only USB port on the Jetson board.

The output of `lsusb` shows:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 2b03:f681  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

But `dmesg -Tw` (the relevant section) shows:

```
[Fri Oct 19 10:32:42 2018] usb 1-2: new full-speed USB device number 12 using xhci-tegra
[Fri Oct 19 10:32:42 2018] usb 1-2: feature bit otg_vbus_off set
[Fri Oct 19 10:32:42 2018] usb 1-2: New USB device found, idVendor=2b03, idProduct=f681
[Fri Oct 19 10:32:42 2018] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Oct 19 10:32:42 2018] usb 1-2: Product: ZED-M Hid Device
[Fri Oct 19 10:32:42 2018] usb 1-2: Manufacturer: STEREOLABS
[Fri Oct 19 10:32:42 2018] usb 1-2: SerialNumber: 10026217
```

Running `" /usr/local/zed/tools/ZED\ Diagnostic"` shows an error on the "Camera" portion of the test, but offers me no resolution. It's clear to me from lsusb that the camera is connecting over USB2.0 and not USB 3.0 - why is that?

I am not running the OpenGL application; I am instead running a simple C++ program that compiles and links just fine:

```
#include <sl/Camera.hpp>

int main(int argc, char **argv) {
    sl::Camera zed;

    sl::InitParameters initParameters;
    initParameters.sdk_verbose = true;
    initParameters.camera_resolution = sl::RESOLUTION_HD720;
    initParameters.depth_mode = sl::DEPTH_MODE_PERFORMANCE;
    initParameters.coordinate_units = sl::UNIT_METER;

    sl::ERROR_CODE err = zed.open(initParameters);

    if (err != sl::SUCCESS) {
        std::cout << "can't open camera" << std::endl;
        std::cout << sl::toString(err) << std::endl;
        zed.close();
        return 1;
    }
}
```

But always outputs the error:

CAMERA DETECTION ISSUE

Right now as it stands, this makes the ZED camera unusable. Any ideas for me?

Myzhar, 2018-10-19:
Have you tried to flip the USB3 cable?

mprat, 2018-10-19:
@Myzhar  I can't believe that worked. Thanks.

closed 2018-10-19 by mprat