stereolabs/zed-sdk #115: file generated with enableRecording() not recognized by ZED Explorer or playback example

jwatte, 2018-11-19:
Using: Zed camera on Jetson Xavier, Ubuntu 18.04.

I captured a SVO file using zed.enableRecording() and putting zed.record() inside my zed.capture() loop. I use HEVC compression.

When trying to open the file in Zed Explorer, or with the "svo recording/playback" sample code, it just complains it's an "invalid SVO file" without any more information (is it too short? bad header? bad image size? bad length? what?)

The header of the file looks very similar to the header of a working file I captured with Zed Explorer:

Broken file header:

    00000000: f5ff ffff 0005 0000 d002 0000 c00a 0000  ................
    00000010: 0000 1041 0000 0000 0000 0000 0000 0000  ...A............
    00000020: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000030: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................

Broken file size:

    -rw-r--r-- 1 nvidia nvidia 5312788 Nov 18 20:33 /opt/xavier/data/20181118-203307/stereo.svo

Working file header:

    00000000: f5ff ffff 0005 0000 d002 0000 c00a 0000  ................
    00000010: 0000 3442 0000 0000 0000 0000 0000 0000  ..4B............
    00000020: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000030: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
    00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................

Working file size:

    -rw-r--r-- 1 nvidia nvidia 218215350 Nov 18 20:18 /opt/xavier/data/HD720_SN2752_20-17-01.svo

The file recorded using Zed Explorer was longer, and also probably didn't use H265 -- can that be a problem?

obraun-sl, 2018-11-19:
It can be a bug with new H265. Is this something that you can reproduce all the time or randomly ?

obraun-sl, 2018-11-20:
Yes, if the file does not properly close, the indexing process that is performed at the end will not be done and the metadata of the file will not be coherent. 
This should be repairable with the repair function but not yet compatible with the SVOEditor --repair command. It will be compatible in next patch version.