stereolabs/zed-sdk #116: cuda error in docker

lyyiangang, 2018-11-21:
I meet following erros in docker, when I am trying to compile depth sensing example.
`root@8a82230b716d:/usr/local/zed/zed-examples/depth sensing/build# ./ZED_Depth_Sensing 
Error: CUDA - OpenGL Interop failed (63)
ERROR: while compiling Shader :
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

ERROR: while compiling vertex shader
ERROR: while compiling Shader :
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

ERROR: while compiling fragment shader
ERROR: while compiling Shader :
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

ERROR: while compiling vertex shader
ERROR: while compiling Shader :
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

ERROR: while compiling fragment shader
Error: CUDA MapResources (33)
Error: CUDA GetMappedPointer (33)
Error: CUDA UnmapResources (33)
Error: CUDA MapResources (33)
Error: CUDA GetMappedPointer (33)
Error: CUDA UnmapResources (33)
Error: CUDA MapResources (33)
Error: CUDA GetMappedPointer (33)
Error: CUDA UnmapResources (33)
Error: CUDA MapResources (33)
Error: CUDA GetMappedPointer (33)
Error: CUDA UnmapResources (33)
Error: CUDA MapResources (33)
`
the docker file I am using is from https://github.com/stereolabs/zed-docker/blob/master/2.7/ubuntu1604/cuda9.0/devel/Dockerfile.

adujardin, 2018-11-21:
Hi,
Could you try with the cuda-gl image instead? 
https://github.com/stereolabs/zed-docker/blob/master/2.7/ubuntu1604/cuda9.0/devel-gl/Dockerfile

lyyiangang, 2018-11-22:
@adujardin cuda-gl version still doesn't work for me. I get following error.
```
root@ac358619370e:/usr/local/zed/tools# ./ZED\ Explorer 
 ** [SVO] Hardware compression (NVENC) not available **
Could not initialize OpenGL
Aborted (core dumped)
```
My GPU is Nvidia GTX 1070m. Here is my OpenGL  information.
```
abc@pc:~/abc$ glxinfo | grep "OpenGL"
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 6.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

```

lyyiangang, 2019-01-18:
I installed graphic card driver like this:'./nv_driver.run -s -no-opengl-files`, and the OpenGL files are not updated. I try to reinstall the driver without '-no-opengl-files`, the cuda-gl docker image works for me now.

closed 2019-01-18 by lyyiangang