stereolabs/zed-sdk #120: ZED SDK, CUDA and download to CPU

BlindElephants, 2018-12-19:
I'm receiving a error when I attempt to call ```download``` on a ```cv::cuda::GpuMat``` that has been bound to sl::Mat data via pointer. The sl::Mat data is being retrieved as GPU (using ```sl::MEM_GPU```).  This is simply for debug, display purposes to visualize what I'm working with and ensure proper functionality. Can advice on this, I realize it might be more of a OpenCV question, although I don't receive any exceptions when doing the same thing elsewhere, not using StereoLabs API. 

Running Ubuntu 18.04, NVIDIA Driver ver.410.78, CUDA 10.0, OpenCV 4.0.

```c++
#include <sl/Camera.hpp>
#include <stdio.h>
#include <stdlib.h>
#include <opencv2/opencv.hpp>

int main(int argc, char **argv)
{
  sl::Camera camera;
  sl::CAMERA_SETTINGS cameraSettings;
  sl::InitParameters initParams;
  sl::RuntimeParameters runParams;

  sl::Mat cameraImage;
  cv::cuda::GpuMat cameraImageCV;
  cv::Mat drawImage;

  initParams.camera_resolution=sl::RESOLUTION_HD720;
  initParams.camera_fps=60;
  initParams.depth_mode=sl::DEPTH_MODE_PERFORMANCE;
  initParams.coordinate_units=sl::UNIT_FOOT;

  runParams.sensing_mode=sl::SENSING_MODE_STANDARD;
  runParams.enable_depth=true;
  runParams.enable_point_cloud=false;

  sl::ERROR_CODE err = camera.open(initParams);
  if (err!=sl::SUCCESS)
  {
    std::cout << sl::toString(err) << std::endl;
    camera.close();
  }
  else
  {
    std::cout << "Camera Opened Successfully" << std::endl;
  }

  cameraImage.alloc(camera.getResolution(), sl::MAT_TYPE_8U_C4);
  cameraImageCV = cv::cuda::GpuMat(cameraImage.getHeight(), cameraImage.getWidth(), CV_8UC4, cameraImage.getPtr<sl::uchar1>(sl::MEM_GPU));

  char key=' ';
  while(key!='q')
  {
    if(camera.grab(runParams)==sl::SUCCESS)
    {
      camera.retrieveImage(cameraImage, sl::VIEW_LEFT, sl::MEM_GPU);
      cameraImageCV.download(drawImage);
      // cv::imshow("Tests", cameraImageCV);
    }
    key=cv::waitKey(10);
  }

  std::cout << "Camera closing" << std::endl;
  camera.close();
  std::cout << "exiting..." << std::endl;

  return 0;
}
```

obraun-sl, 2018-12-20:
Hi,
By Default, the sl::Mat::alloc() function will create the CPU memory only. Therefore, when you create your gpuMat, the gpu buffer ptr is not valid (null). Therefore they do not share the same pointer.

I think you just need to change the line : 
` cameraImage.alloc(camera.getResolution(), sl::MAT_TYPE_8U_C4);`
to 
` cameraImage.alloc(camera.getResolution(), sl::MAT_TYPE_8U_C4,sl::MEM_GPU);`

and this should work.

BlindElephants, 2018-12-20:
Works perfectly. Thanks for that pointer.

closed 2018-12-20 by BlindElephants