stereolabs/zed-sdk #122: Positional tracking recovery issue

ABERNARD1996, 2019-01-25:
Hello,
I plan on using the Zed mini for indoor positional tracking with the Nvidia Jetson TX2. I'm currently getting familiar with the SDK and samples provided (thanks for your work there BTW), but I have a problem with the positional tracking ; when the tracking state is TRACKING_STATE_SEARCHING, it very very rarely finds its position back, even if we spend a good amount of time not moving the camera a lot and scanning its surroundings. 
Does anyone have any advice on how to avoid this? 
Thanks a lot.

JosephRedfern, 2019-07-12:
@ABERNARD1996 Did you get anywhere with this? Currently have the same issue. Spent a good amount of time and effort yesterday recording lots of data, but cannot process *any* of it right now, as positional tracking isn't working. Have tried processing the file on 3 different machines with two versions of the SDK (2.8 and 2.7.1) with no success. Very frustrating that the SDK doesn't output any more information to help debug this.

adujardin, 2019-07-15:
When the Tracking state is "Searching", it means the positional tracking is lost. The initial reference frame is not found and the module tries to find an older position from a database (see [documentation]( https://www.stereolabs.com/docs/positional-tracking/spatial-memory/)) recorded on the fly ([SLAM](https://en.wikipedia.org/wiki/Simultaneous_localization_and_mapping)).

If  [`TrackingParameters::set_floor_as_origin`](https://www.stereolabs.com/docs/api/structsl_1_1TrackingParameters.html) is enabled, the state will be "searching" from the start, and the tracking will begin once the floor plane is found.
By default this feature is disabled. In that case, if the camera is lost after only a few frames, it will be difficult to relocate from only a tiny database, the tracking should be restarted.

If the tracking is lost after quite some time, the best way to find back a known reference is to return to previously visited positions at approximately the same viewpoints. 

In pure exploration, when the camera is only moving in a previously unseen environment, the recovery won't be possible.

liangty10, 2019-10-16:
@adujardin Hi I also met this problem. Even if I enabled set_floor_as origin, tracking lost still happens, and it's very hard to recover. I have experienced ZED World app for mixed reality, the virtual objects were steady. Can you offer any advices?

closed 2022-04-26 by github-actions[bot]