stereolabs/zed-sdk #125: CMake failed the configuration

ZHOUXIAOYI95, 2019-02-11:
Hi, I'm a beginner. I followed each step of the tutorial instruction, but the Cmake cannot configure the project file and the warning shows ' error in the configuration process, project file may be invalid'. 

Please offer me any ideal to solve it. Many thanks!
[image]

obraun-sl, 2019-02-11:
Hi,
You need to install visual studio 2015 or 2017 (C++) to be able to build examples. 
OB

ZHOUXIAOYI95, 2019-02-11:
> Hi,
> You need to install visual studio 2015 or 2017 (C++) to be able to build examples.
> OB

hi, problem solved, many thanks.

closed 2019-02-11 by ZHOUXIAOYI95