stereolabs/zed-sdk #126: Record SVO without CUDA

shehzaman, 2019-02-11:
Hi, 

Is it possible to record in .svo format without CUDA? I have plenty of field work and it would help if I could record and send .svo files over to my server with CUDA to process depth. The heavy lifting will be done server side. 

Any way to achieve this? Thanks!

obraun-sl, 2019-02-11:
Hi,
You can use ZED Explorer tool to record SVO without NVIDIA card (this tool does not require NVIDIA graphic card). However, you won't have access to AVCHD/HEVC compression mode.

Best,
OB

shehzaman, 2019-02-11:
That should be fine, I would like to keep as much raw data as possible. 

Is it possible to write some C++ executable for this? I would like to fix the resolution and record so that there is minimum effort from anyone using it on field.

obraun-sl, 2019-02-12:
Currently the C++ code will be linked to CUDA (ZED SVO Recording) and I assume this is not what you want.

You can use ZED Explorer in cmd line to start a record and choose resolution/FPS. For example : 

./ZED\ Explorer -r HD1080 -f 15 --cpm 0 -o output.svo will record a SVO at HD1080@15fps in Lossless
You can also specify the number of frame in the svo with --l option. You can then put that call in a script.

See ./ZED\ Explorer -h for more options

Best,
OB

shehzaman, 2019-02-12:
Thank you for that solution! I will definitely look at the options more closely.

Would ZED SDK install on a laptop without an Nvidia GPU? The only way I can get ZED Explorer is with the SDK and all documentation shows that CUDA is required. 

Any way to install ZED Explorer as a standalone application? I was looking to compile for Linux, Windows and MacOS systems. 

Once again, thank you so much for all the help!

obraun-sl, 2019-02-12:
Yes you can install it without NVIDIA. It will just tell you that CUDA is missing but you can ignore that.
This way is better than just copy the application.

shehzaman, 2019-02-12:
Great thank you!

shehzaman, 2019-02-19:
Reopening the issue as ZED Explorer filed to open on a new Windows 10 laptop. 

I received the following errors:

* Missing mfc140u.dll
* Missing VCOMP140.DLL

Checked for these errors online, and they all point to Visual Studio - do I need to install the the entire Visual Studio package? Is there a minimal install to support ZED Explorer? 

It would be great if I could write a standalone C++ application for SVO recording. 

[image]

obraun-sl, 2019-02-19:
You can just install the VC redist from here : https://www.microsoft.com/en-us/download/details.aspx?id=48145

Did you install the ZED SDK as mentionned? because this redist should be installed by the SDK itself.

shehzaman, 2019-02-19:
Ah you are correct! I cancelled the Visual Studio install and CUDA 10 install during ZED SDK installation. 

Thought that entire Visual Studio 2015 was being installed, could not afford that space on disk. 

Ran the SDK again, let Visual Studio 2015 redist install (around 13mb) and it works!

My apologies for this mistake. Thank you so much for your help!

closed 2019-02-19 by shehzaman